dgschwend/zynqnet #1: availability of your MS thesis

kaishijeng, 2016-07-28:
Hi,

  Is your thesis publicly available? If yes, can you show me a download link?

Thanks,

dgschwend, 2016-07-28:
Hi kaishijeng

Thanks for your interest. I'm currently finishing the written report, and I plan to upload it to Github towards the middle of August.
I'll add a corresponding note to the README, didn't think anyone would stumble upon the repository so soon... :)

Best regards,
David

kaishijeng, 2016-07-28:
David

   Thanks for an update

kaishijeng, 2016-08-23:
Do we still plan to release the report in this month?

Thanks

dgschwend, 2016-10-14:
Hi @kaishijeng. I finally got all the necessary permissions for publication. You can find the whole Master Thesis including all source code, generated files, as well as the Project Report in the repository.

closed 2016-10-14 by dgschwend

kaishijeng, 2016-10-16:
David

  Thanks for letting me know your report is available.

FC