dgschwend/zynqnet #2: When I run the project ,I found the top5 result is change.Why?And I also find the timing is slack when the ipcore running 200Mhz,Slack is -3ns

guodapengzte, 2016-12-07:
the first:
Result (top-5):
====================
    31.17%: class 238 (output  10.26)
    26.41%: class 264 (output  10.10)
    11.45%: class 156 (output   9.26)
     3.58%: class 232 (output   8.10)
     2.90%: class 218 (output   7.89)

TestBench Result: FAILURE
Actual: 31.17, Expected: 88.38

the second:
Result (top-5):
====================
     0.75%: class 346 (output   2.61)
     0.49%: class 711 (output   2.19)
     0.49%: class   9 (output   2.19)
     0.47%: class 727 (output   2.14)
     0.47%: class  25 (output   2.14)

TestBench Result: FAILURE
Actual:  0.75, Expected: 88.38
the third:
Result (top-5):
====================
    15.67%: class 148 (output   6.63)
     9.07%: class 548 (output   6.09)
     4.59%: class 598 (output   5.41)
     2.51%: class 782 (output   4.80)
     2.03%: class 904 (output   4.59)

TestBench Result: FAILURE
Actual: 15.67, Expected: 88.38
root@localhost:~/tmp#

guodapengzte, 2016-12-13:
The problerm has solved. 
first:  Before you download your fpga.bit ,you must set the cpu hp port to 32 bit and the fpga0_clk_ctrl0 register：
the method （reference as ug585）: 
devmem2 w 0xf8008000 1
devmem2 w 0xf8008014 1
this is setting the HP PORT from 64bit to 32 bit 
the same method to set the fpga0_clk

dgschwend, 2016-12-13:
Perfect, I'm glad you finally found the solution. It's important to set
these registers *before* programming the .bit file to the FPGA.

guodapengzte, 2016-12-14:
Hi David:I want to use difference pic to verify the result,and Iwant  directly to use yours train model:So, First,I think you must have a excel to list that class 1    chass 2 ...and so on to point dog or cat .For example:   I change your pic a dog with myself a dog , and affter runing I get a difference result ,but I do'nt known the result is right or wiong.
So, can you share your the excel .BestWishes

dgschwend/zynqnet <zynqnet@noreply.github.com>抄　送：DaPeng Guo <guo.dapeng@aliyun.com>; Author <author@noreply.github.com>主　题：Re: [dgschwend/zynqnet] When I run the project ,I found the top5 result is change.Why?And I also find the timing is slack when the ipcore running 200Mhz,Slack is -3ns (#2)

dgschwend, 2016-12-14:
Hi! The mapping is defined in `synset_words.txt` from the ImageNet database for Caffe (see http://caffe.berkeleyvision.org/gathered/examples/imagenet.html). You can grab a copy by running Caffe's `./data/ilsvrc12/get_ilsvrc_aux.sh` or from here: https://github.com/happynear/caffe-windows/blob/master/examples/GoogLeNet/synset_words.txt
The line numbers (starting from 0) correspond to ZynqNet class outputs.

guodapengzte, 2016-12-15:
Thank you very much.
Next,I will train the network by myself picture,and then I will share the result.

dgschwend, 2016-12-20:
Attaching previous e-mail exchange with @guodapengzte:

When I run the project in the Fpga board ,I find some error(I compile the C++code under the _FIRMWARE with the Xilinx SDK).
The Fpga bit is yours:  zynqnet_200MHz.bit ( my own fpga.bit also has the same problem)

Result (top-5):
====================
     0.56%: class 711 (output   2.19)
     0.56%: class   9 (output   2.19)
     0.53%: class 727 (output   2.14)
     0.53%: class  25 (output   2.14)
     0.48%: class 597 (output   2.03)

TestBench Result: FAILURE
Actual:  0.56, Expected: 88.38

**Solution**:
Probably, your AXI HP bus is configured for 64-bit. You need to configure it for 32-bit width, otherwise you get the wrong classification results, and very low accuracies for all classes. The bus width can be configured in the Vivado Design Suite, Block Designer.

guodapengzte, 2016-12-22:
Hi David :      Now ,I want to change the *SHARED_DRAM from PS_DDR to PL_DDR.Because the ps_ddr is always used by other function.     And,now I change the *SHARED_DRAM address to point to PL_DDR.And,I find a error log :     test_ddr.elf: ../src/xfpga.cpp:56: void XFPGA_setDRAMBase(): Assertion `SHARED_DRAM_64b - (u32)SHARED_DRAM_64b == 0' failed.          It means that the most high bit of SHARED_DRAM's address cann't set to 1 .Why ?    And,is there others place to change?         Best Wishes

dgschwend, 2016-12-23:
Hi guodapengzte
I haven't tried using PL_DDR. It seems that your physical address on the AXI bus is then >32 bits.
You should check that all variables and functions which hold/return memory addresses (e.g. SHARED_DRAM_BASE_ADDR, SHARED_DRAM_virtual(), SHARED_DRAM_physical()) can handle 64-bit integers.
Then, you need to find out how you can point the SHARED_DRAM memory port in the HLS code to a 64-bit address
`#pragma HLS INTERFACE m_axi depth = DRAM_DEPTH port = SHARED_DRAM offset = slave bundle = memorybus register`
--> here, `offset = slave` creates a AXI-Lite-accessible register _SHARED\_DRAM_ which holds the base address of the external DRAM. This register is currently only 32 bit, I think. You can see this in `xfpga_hw.hpp`, in XFPGA_TOP_AXILITE_BITS_SHARED_DRAM_DATA = 32. Presumably because the whole AXI-Lite bundle `axilite` is only 32-bit. I don't know if you can increase AXILITE bus width.
Alternatively, you would need to create your own SHARED_DRAM base-address register (64-bit), write this register via AXILITE, and then add this base-address offset to every AXI-Master access on the SHARED_DRAM port.
So, it's quite a lot of work to change. Alternatively, would it be possible to map the PL_DDR to a lower memory address?
Regards
David

divyapraneetha, 2017-02-28:
Hi David,

I am working on a similar project. Modifying the network according to the FPGA resources I have.  I would like to know the input image format being followed. 

I have gone through changing image files to lmdb sets. but the code you have implemented takes .bin image.

I would like to know how did you define input formats.

Any help in this is highly appreciable.

Thanks
Divya

dgschwend, 2017-03-01:
Hi Divya, I opened another issue for your question, see https://github.com/dgschwend/zynqnet/issues/4
Regards
David

closed 2017-03-01 by dgschwend