dgschwend/zynqnet #8: Final caffemodel file

canhld94, 2017-05-13:
Hi David,
Can I have final caffemodel of ZynqNet? Your thesis is amazing, but FPGA version of ZynqNet have not been verified completely. I've made some tests on ZynqNet (FPGA version, running on OZ745 Kit) and there're some interesting results. First, with 5000 image from ILSVRC2012 validation set, top 1 result is 54% and top-5 is 77%. Second, I config my FPGA clock at > 200MHz (it's easy, just config a system register by devmem command), but it still take 1.7s for 1 image. And the last is the result of FPGA version and software version (C++ calssification, caffe framework) is not the same within 1 image. So i think the caffemodel file you uploaded is not the same with the file weighs.bin.
About the second issue, I've read your code quite carefully, and I think the bottleneck is memory (DRAM) access, so even you clock at double clock rate, performance will not be doubled.
Anyway, you've make a great framework, a lot of people are grateful to you, including me 😄

dgschwend, 2017-05-18:
Hi canhld94
Thanks for your interest and the comments, and congratulations on getting the whole thing running.
You can find the final .caffemodel file here: https://github.com/dgschwend/zynqnet/blob/master/_TRAINED_MODEL/snapshot_iter_300280.caffemodel I'm fairly certain this is the model from which I derived the weights.bin.
Your observations are quite interesting. I never did any large-scale verification of the accuracy on the FPGA because I was sure the calculations were bit-identical to the Caffe model, and my few test images yielded the exact same inference results in Caffe and on FPGA. Have you tried running the verification at 100 MHz to make sure this is not a side-effect from running at a too high clock rate?
The limited throughput gain from the higher clock rate is also rather unexpected. I am sure that I checked the bandwidth of the memory interfaces and it was definitely far away from being a bottleneck... just very roughly: 10MB parameters + 40MB activations in 1 second = 50 MB/s, which is close to negligible. And accesses are almost perfectly linear, hence data can be transferred in bursts. There must be another problem...
Hope you can find out more, get back to me if I can help you with anything else!

canhld94, 2017-05-24:
Hi David,
I also did a test in python (caffe), and with all 50000 image from  ILSVRC2012 validation set, top 1 result is 58,27% and top-5 is 81,46%. And FPGA's result doesn't change, 54% and 77% at 153.33 MHZ. I think it's all I can do to verify your ZynqNet. It's sound bad, right 😄. It's must be some differences between our data set, or our method in verifying the network. 
One more thing is Zynq 7000 is definately not a good platform for CNNs. It's not powerful enough (even xc7z100). Do you intend to bring ZynqNet to stronger devices, like ultrascale or ultrascale+?
And finally thank you a lot. I've used your ZynqNet and as core-processing algorithm in my Bsc graduation thesis without re-tranning or modification. I should did it better, but I only have 3 months to do everything (3 month ago i even didn't know there a thing called CNN in this world 😄). I'm gonna finish my thesis and I won't do any other tests in next 2 weeks. If you are interesed, I will send you all materials I've done.
Thanks,

dgschwend, 2017-05-24:
Here's a screenshot from my training history in DIGITS:
[image]
Did you use the image transformations as shown in https://github.com/dgschwend/zynqnet/issues/4 ? If you don't crop-and-fill the images similar to the training, this may degrade accuracy.
Regarding the FPGA size: it depends. For image classification / detection / segmentation of large images with many classes, I agree that Zynq7000 devices are probably not an ideal choice. However, especially for smaller tasks and when an FPGA is already available in the system (which is the case in many computer vision processing systems), you can certainly run a CNN on FPGA.
Porting to larger FPGAs (especially if it's also a Zynq/SoC) should not be much of a problem, but I have currently no plans doing it. I would wait for frameworks/libraries from the FPGA vendors (e.g. hinted at during Xilinx presentation https://www.youtube.com/watch?v=05dozOTzwhA).
Glad to see my thesis has been of use for you! Good luck with the thesis. :+1:

canhld94, 2017-05-24:
I follow the instruction here http://nbviewer.jupyter.org/github/BVLC/caffe/blob/master/examples/00-classification.ipynb . I've also read your thesis, you cropped 256x256 pixel from a 300x300 image, but as I see in Imagnet validation set, the resolution of all image is 350x500 or approximately this value. So may be our dataset are not the same?

dgschwend, 2017-05-24:
No, our dataset is not going to be the same if you don't prepare the input images as indicated in the issue mentioned above. ;)

––
David Gschwend • +41 79 552 74 00 • Winkelriedstr. 56 • 6003 Luzern

closed 2017-06-02 by dgschwend