dgschwend/zynqnet #11: HLS malloc function question

liwei1018, 2017-05-26:
Hi David,

Below is the screenshot of HLS runs. Please ensure the files in TestBench and src are correctly placed. When I tried for C/RTL Cosimulation, I got error message attached below.
[image]

Thanks

dgschwend, 2017-05-26:
Looks good to me... have you changed anything in the source code? Can you try with Vivado 2016.2 instead, which is the version I used? Maybe Xilinx changed something... Can you compile + run the C simulation alone?

liwei1018, 2017-05-29:
Hi David,
 thank you for your help
 I try to used Vivado 2016.2 to  run the C simulation . I still I have such a mistake.
[image]
[image]
this is alloc memory region. Is this err and alloc memory size related?
thanks

canhld94, 2017-05-29:
In line 199 file cpu_top.cpp, you'll see a memory allocation like this: SHARED_DRAM = (char *)malloc(total_size);. What you need to do is adding more memory, something like SHARED_DRAM = (char *)malloc(2*total_size); or bigger if you want. I'm not yet figured out where's the problem but you can use it as a temporary solution.

liwei1018, 2017-05-30:
hi canhld94, dgschwend
 thank you for your help. I solve this err with your help. but i have a new err about vivado Diagram 
[image]
Can you tell me what 's the mistakes. how to solve it.
thanks

dgschwend, 2017-06-02:
It looks like Vivado could not find the floating-point IP... I don't understand why.
Did you use the same version for Vivado HLS and Vivado Design Suite?

liwei1018, 2017-06-02:
thank you a lot
 I solved the problem with crate fpga_top IP use VHDL. floating-point IP is VHDL file so  Vivado could not find the floating-point IP . if crate fpga_top IP use Verilog.
thanks

dgschwend, 2017-06-02:
Great, glad you figured it out!

closed 2017-06-02 by dgschwend