dgschwend/zynqnet #12: How to get weights.bin

Brillise, 2017-05-29:
Hi David,
Thank you so much for your ZyngNet and your thesis! I helps me a lot to learn CNN~^o^~
I'm interested in your framework. And I have realized your network on FPGA using weights you provided(weights.bin). Now I want to try some other networks e.g. VGG, so how can I convert the original .caffemodel file to the .bin file, which can be read by your code. I think you do something for transforming snapshot_iter_300280.caffemodel to weights.bin.

Thanks,

dgschwend, 2017-06-02:
Hi Brillise
You can convert Caffe networks (.prototxt + .caffemodel files) to ZynqNet-compatible networks (weights.bin, network.hpp/cpp) by using the scripts in [tools/convert_caffemodel](https://github.com/dgschwend/zynqnet/tree/master/tools/convert_caffemodel). See the readme file in the folder.
But beware that the conversion script is very simple and targeted at topologies similar to SqueezeNet/ZynqNet. Also, not all layers and topologies supported by Caffe are also supported by ZynqNet. You will probably have to do quite a lot of manual work to get VGG or a similar network running (e.g. convolutions >3x3, pooling, fully-connected layers not supported).
Regards
David

Brillise, 2017-06-02:
Hi David,
Thanks for your reply!
Actually, I even haven't installed Caffe now. We try to do everything independent of Caffe. After reading running requirements of your scripts, it seems that I must install one 〒_〒

arashazizima, 2017-06-09:
Hi David,
If you remember, I already contacted you.
I tried to reuse your code in some parts of a project. However, I have the problem when I synthesize the code on the FPGA by Vivado HLS. The problem is related to the DSP utilization. I reuse the following code as the dot-product in my convolutional engine. However, the DSP utilization is much lower than the value I expected. Also, I receive this warning message: "WARNING: [SYN 201-303] Cannot apply functional unit assignment of 'FAddSub_fulldsp'".

Could You please help me with that based on your experience?  how can I generally control the resource allocation in the Vivado HLS other than the compiler directives?     
Thanks,
Arash

void CAccelerator::vector_dot_product(const data_t in[Tn], const data_t w[Tn], data_t& result)
{
#pragma HLS INLINE

data_t accumulator = 0.0f;
#pragma HLS RESOURCE variable=accumulator core=FAddSub_fulldsp

data_t multresult[Tn];
#pragma HLS ARRAY_PARTITION variable = multresult complete dim = 0
#pragma HLS RESOURCE variable=multresult core=FAddSub_fulldsp

for(int h=0; h<Tn; h++)
#pragma HLS UNROLL factor=Tn
multresult[h]=in[h] +w[h];

for(int h=0; h<Tn; h++)
#pragma HLS UNROLL factor=Tn
accumulator = accumulator + multresult[h];

result = accumulator;
}

dgschwend, 2017-06-09:
@arashazizima Could you please open a new issue? Your question has nothing to do with weight conversion...

dgschwend, 2017-06-09:
@Brillise Actually, it's possible to avoid Caffe. You can parse the .prototxt and .caffemodel files directly using Google Protobuf (https://github.com/google/protobuf) or your own custom parser. But it involves some extra effort, and you still have the dependency on Google Protobuf... but at least you no longer need the whole Caffe installation. ;)

closed 2017-06-09 by dgschwend