dgschwend/zynqnet #13: Underutilization of DSP resources

arashazizima, 2017-06-09:
Hi David,
I tried to reuse your code in some parts of a project. However, I have the problem when I synthesize the code on the FPGA by Vivado HLS. The problem is related to the DSP utilization. I reuse the following code as the dot-product in my convolutional engine. However, the DSP utilization is much lower than the value I expected. Also, I receive this warning message: "WARNING: [SYN 201-303] Cannot apply functional unit assignment of 'FAddSub_fulldsp'".

Could You please help me with that based on your experience? how can I generally control the resource allocation in the Vivado HLS other than the compiler directives?
Thanks,
Arash

void CAccelerator::vector_dot_product(const data_t in[Tn], const data_t w[Tn], data_t& result)
{
#pragma HLS INLINE

data_t accumulator = 0.0f;
#pragma HLS RESOURCE variable=accumulator core=FAddSub_fulldsp

data_t multresult[Tn];
#pragma HLS ARRAY_PARTITION variable = multresult complete dim = 0
#pragma HLS RESOURCE variable=multresult core=FAddSub_fulldsp

for(int h=0; h<Tn; h++)
#pragma HLS UNROLL factor=Tn
multresult[h]=in[h] +w[h];

for(int h=0; h<Tn; h++)
#pragma HLS UNROLL factor=Tn
accumulator = accumulator + multresult[h];

result = accumulator;
}

dgschwend, 2017-06-09:
What type is data_t? Why do you sum up in[h] and w[h] vectors, I would expect multiplication for a dot-product?
You can only control the resource allocation using pragmas or influence the resource which is used by writing your code differently.

arashazizima, 2017-06-09:
I have "typedef float data_t" for the data_t. 
it is a **vector**-dot-product. For vector-dot-product, after multiplication, the partial multiplications should be sum up.

dgschwend, 2017-06-09:
Alright, so regarding functionality, the function you're computing is `result = sum(w) + sum(in)`, where `in` and `w` are float vectors.
Regarding implementation, you partition "multresult" completely, so you get individual registers for all partial results (in0+w0, in1+w1, ...). You unroll the summation loops, so you get parallel computation units. Your block will have Tn*32 input wires for "in", Tn*32 input wires for "w", Tn adder cores computing the vector "multresult" (w0+in0, ...) of partial sums, storing them into Tn individual 32bit registers, and then a chain (or possibly a tree, if the compiler is clever enough) of adders to compute the "accumulator"/result. Is that what you expect?Thinking about it, your problem presumably comes from the "pragma HLS RESOURCE variable=multresult ...". Multresult is an array, and what you actually want to say is that the operation on *every array element* should be executed using the AddSub_fullDSP resource. You might try "pragma HLS RESOURCE variable=multresult[0] ..." and so on. Or you might need to find another way to specify this resource...

dgschwend, 2017-06-09:
If the pragma syntax for the array elements does not work, you could try it with
set_directive_resource -core AddSub_fullDSP vector_dot_product multresult[0]
or similar... see e.g. https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/HLS-Efficiently-apply-RESOURCE-constraints-to-expressions/td-p/355287
In summary, the syntax for allocating resources is not very flexible and powerful. You might need to rewrite your code. E.g. Maybe you could use a temporary local variable within the loop to do the addition, and allocate the resource for this variable/operation, and then unroll the loop?

arashazizima, 2017-06-12:
Thank you for your comments. First, "multresult[h]=in[h] +w[h];" should be changed to "multresult[h]=in[h] *w[h];" for vector-dot-product. Probably, I changed it for some testing and I forgot to correct it when I copied the code here. Your interpretation of the code is the functionality that exactly I am trying to implement. it is the functionality that I have in my mind and I want to implement the addition and multiplication with DSP resources of the FPGA.
I did all of your suggestions, but still, I cannot get expected DSP utilization. For example, I modified the code as follows, this time there is no any warnings, but I got zero DSP utilization!!

**data_t temp;
#pragma HLS RESOURCE variable=temp core=FMul_fulldsp
 for(int h=0; h<Tn; h++)
#pragma HLS UNROLL factor=7
{
temp=in[h] * w[h];
multresult[h]=temp;
}**

I am totally confused. I am using a free version of Vivado HLS which is available in the blow link.
https://www.xilinx.com/member/forms/download/xef.html?filename=Xilinx_Vivado_SDK_2017.1_0415_1_Win64.exe&akdm=0
 
Is it possible that problem is with the current version of Vivado which I am using?? Because when I change the devices I get below warning. 

[image]

closed 2017-08-08 by dgschwend