dgschwend/zynqnet #14: Latency and power consumption of each function

thenamnguyen, 2017-07-21:
Hi David,
I'm try to determine latency and power consumption of each function, such as: ImageCache::preloadPixelFromDRAM, WeightsCache::loadFromDRAM... when it run on FPGA. I try to synthesis these functions on Vivado HLS to determine latency, but the report show that latency is 0. I think it isn't correct. Do you have any idea for problem? thank you so much.

dgschwend, 2017-07-22:
Hi thenamnguyen

I think the latency, throughput and power for loads from DRAM cannot be estimated with Vidvado HLS. Vivado HLS only considers the memory port as a connection to some memory bus, but the memory bus itself is out of scope.
As a matter of fact, you can actually connect the finished IP to different memory ports in Vivado Block Designer (eg. via ACP and processor cache or directly to PL DRAM or directly to PS DRAM). All these memories will have different throughputs, latencies and power requirements. Your best guess would probably be to study the datasheets, or to do benchmarks (measure clock cycles between DRAM request and DRAM response, measure power consumption with idle bitstream, and with constant memory read/write). The memory accesses to DRAM from Zynqnet accelerator are almost completely linear and can thus be bursted. Therefore latency will only be relevant for the first transfer of each burst, after that it's primarily throughput which is relevant. To minimize this latency, connect the IP to the ACP port to profit from the CPU cache...

I hope this helps, let me know how it goes!

Best regards
David

thenamnguyen, 2017-07-24:
Thanks for your response. I have a question. I run Zynqnet on FPGA, and how do I know when a function starts and when it stops? Can I do anything on HLS code to know that?

dgschwend, 2017-07-24:
Hi 

This is not possible by default. You could use Chipscope to look at signals in the running FPGA. Probably you would need to add some signals to the top-level block to show when the function of interest starts and stops. You could also route these signals to a physical pin of the FPGA, and use an oscilloscope to measure the timings.

Best regards
David

thenamnguyen, 2017-07-25:
Hi,
I suppose that I have a top function . Inside top function, it call A function, and I want to determine active time of A. I try to place checkpoint into top function, and checkpoint is output signal. Before start function  A, checkpoint = 1, and after stop function A, checkpoint = 0. when I synthesis, I realize that checkpoint is valid when top function return and value of checkpoint is 0. I want to have a output signal, when function A starts its value is 1, and when function A stops its value is 0. what should I do with HLS code? Thank you so much!

dgschwend, 2017-08-08:
I don't actually have experience with this, but you might want to take a look at sections "Interface Synthesis and Multi-Access Pointers" (p. 94) and "Specifying Manual Interface" (p. 104) and "Understanding Volatile Data" (p. 355) in the HLS Guide https://www.xilinx.com/support/documentation/sw_manuals/xilinx2017_1/ug902-vivado-high-level-synthesis.pdf

closed 2017-12-12 by dgschwend