dgschwend/zynqnet #15: Zynqnet problem

RoxanaMir, 2017-07-21:
Dear David
I created a new project in Vivado HLS and imported the “_HLS_CODE” files in the directory for source files. After synthesis, I met the following error:
 
Zip/zynqnet-master/_HLS_CODE/unittests.cpp:50:23: error: no member named 'setiosflags' in namespace 'std'
    std::cout << std::setiosflags(std::ios::fixed) << std::setprecision(2)
                 ~~~~~^
Zip/zynqnet-master/_HLS_CODE/unittests.cpp:50:60: error: no member named 'setprecision' in namespace 'std'
    std::cout << std::setiosflags(std::ios::fixed) << std::setprecision(2)
 
 I will be grateful if you help me and guide me to a successful synthesis

Best regards,
Roxana

------
Roxana Mir
MSc student,
CE department
Science and research branch of IAU

dgschwend, 2017-07-22:
Hi Roxana

You can simply remove the line from the source code, its only used to set the formatting for the debug output. That's not necessary and the tests should work anyway.

Regards
David

––
David Gschwend • +41 79 552 74 00 • Winkelriedstr. 56 • 6003 Luzern

RoxanaMir, 2017-07-23:
Dear David, 
I will do it as you say. 

Thank you.

RoxanaMir, 2017-08-02:
Hi David
I did as you say and the problem was over but now I met this error :
"Too many positional arguments specified"

I'm using Vivado 2015.4 , may be any problem with this version?
I will be thankful for your help.

Best regards,
Roxana

dgschwend, 2017-08-02:
Can you give the entire error message? I need to know where the problem occurs. This does not sound like an issue in the C code.
Trying a different version of Vivado might be a good idea as well.

RoxanaMir, 2017-08-02:
Thanks for your help, the full error was as below:

@E [HLS-101] 'add_files': Too many positional arguments specified.

FORMAT
  add_files [OPTIONS] <src_files>
     src_files <string>  (required)
    -cflags <string>
    -tb

command 'add_files' returned error code
    while executing
"add_files OK Zynqnet Master Complete/zynqnet-master/_HLS_CODE/vivado_include/ap_fixed.h"
    (file "C:/Users/Asus/Desktop/ZynqNet/solution1/csynth.tcl" line 8)
    invoked from within
"source C:/Users/Asus/Desktop/ZynqNet/solution1/csynth.tcl"
    invoked from within
"hls::main C:/Users/Asus/Desktop/ZynqNet/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"

And can you tell me which file has to be added as "Test Bench"?

dgschwend, 2017-08-02:
> src_files (required)

You need to save the files on a path without spaces (e.g. C:\zynqnet-master\ instead of "OK Zynqnet Master Complete/zynqnet-master").

The TB consists of: cpu_top.*, indata.bin, weights.bin, unittests.* (iirc.)

RoxanaMir, 2017-08-08:
Hi David,
Thanks for helping me and my silly mistakes. I corrected the file paths and TBs, but now I'm met a bad error, I had error on "printf" and "fopen". It doesn't recognize them and the following error is reported:

@E [HLS-70] In file included from C:/Users/Asus/Desktop/zynq/solution1/.autopilot/db/netconfig.pragma.1.cpp:1:
In file included from ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:1:
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:103:22: error: use of undeclared identifier 'fopen'
  FILE *filehandle = fopen(filename, "rb");
                     ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:105:5: error: use of undeclared identifier 'printf'
    printf("ERROR: File %s could not be opened!\n", filename);
    ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:120:5: error: use of undeclared identifier 'fread'
    fread(weights_addr, sizeof(data_t), num_weights, filehandle);
    ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:123:3: error: use of undeclared identifier 'fclose'; did you mean 'false'?
  fclose(filehandle);
  ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:183:3: error: use of undeclared identifier 'printf'
  printf("%6s: %3dx%-3d x %3d > %-3d, ", layer->name, (int)layer->height,
  ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:186:3: error: use of undeclared identifier 'printf'
  printf("CONV (%dx%d)/%d%s %s, ", (int)layer->kernel, (int)layer->kernel,
  ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:190:3: error: use of undeclared identifier 'printf'
  printf("IN @mem(%8lu-%8lu" "" "B), ",
  ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:194:3: error: use of undeclared identifier 'printf'
  printf("OUT @mem(%8lu" "" "B), ",
  ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:197:3: error: use of undeclared identifier 'printf'
  printf("WEIGHTS @mem(%8lu-%8lu" "" "B)",
  ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:202:5: error: use of undeclared identifier 'printf'
    printf(" (split%d)", (layer->is_first_split_layer) ? 1 : 2);
    ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:205:5: error: use of undeclared identifier 'printf'
    printf(" GLOBAL POOL");
    ^
ZynqnetMasterComplete/zynqnet-master/_HLS_CODE/netconfig.cpp:207:3: error: use of undeclared identifier 'printf'
  printf("\n");
  ^
12 errors generated.

Please help me with this error.
Best Regards

dgschwend, 2017-08-11:
Please remove "netconfig.cpp" from synthesis, this file is only needed for the testbench. "netconfig.hpp" is required for both synthesis and testbench.

Here is a complete list of the required files and whether they are intended for TB or synthesis (see attribute "tb"):

<file name="../../weights.bin" sc="0" tb="1" cflags=" "/>
<file name="../../unittests.hpp" sc="0" tb="1" cflags=" "/>
<file name="../../unittests.cpp" sc="0" tb="1" cflags=" "/>
<file name="../../network.hpp" sc="0" tb="1" cflags=" "/>
<file name="../../network.cpp" sc="0" tb="1" cflags=" "/>
<file name="../../netconfig.hpp" sc="0" tb="1" cflags=" "/>
<file name="../../netconfig.cpp" sc="0" tb="1" cflags=" "/>
<file name="../../indata.bin" sc="0" tb="1" cflags=" "/>
<file name="../../cpu_top.hpp" sc="0" tb="1" cflags=" "/>
<file name="../../cpu_top.cpp" sc="0" tb="1" cflags=" "/>
<file name="fpga_top.cpp" sc="0" tb="false" cflags=""/>
<file name="fpga_top.hpp" sc="0" tb="false" cflags=""/>
<file name="gpool_cache.cpp" sc="0" tb="false" cflags=""/>
<file name="gpool_cache.hpp" sc="0" tb="false" cflags=""/>
<file name="image_cache.cpp" sc="0" tb="false" cflags=""/>
<file name="image_cache.hpp" sc="0" tb="false" cflags=""/>
<file name="memory_controller.cpp" sc="0" tb="false" cflags=""/>
<file name="memory_controller.hpp" sc="0" tb="false" cflags=""/>
<file name="netconfig.hpp" sc="0" tb="false" cflags=""/>
<file name="network.hpp" sc="0" tb="false" cflags=""/>
<file name="output_cache.cpp" sc="0" tb="false" cflags=""/>
<file name="output_cache.hpp" sc="0" tb="false" cflags=""/>
<file name="processing_element.cpp" sc="0" tb="false" cflags=""/>
<file name="processing_element.hpp" sc="0" tb="false" cflags=""/>
<file name="weights_cache.cpp" sc="0" tb="false" cflags=""/>
<file name="weights_cache.hpp" sc="0" tb="false" cflags=""/>

RoxanaMir, 2017-08-12:
Thank you very much David, I really appreciate your help.

Best regards,
Roxana

RoxanaMir, 2017-08-20:
Dear David,
I added the source files and test bench files as you said and also I added the "Vivado Includes" by the following path: Properties/GCC C++ Compiler/Includes/Include files.
After Synthesis, I met the following error:

../_HLS_CODE/cpu_top.cpp:103:10: error: cast from pointer to smaller type 'long' loses information

The error is for:
LOG("SHARED_DRAM is at address: %lu\n", (long)SHARED_DRAM);
    int weights_offset =
        ((long)SHARED_DRAM_WEIGHTS - (long)SHARED_DRAM) / sizeof(data_t);
    int input_offset =
        ((long)SHARED_DRAM_DATA - (long)SHARED_DRAM) / sizeof(data_t);

Can you help me again?

Thanks

dgschwend, 2017-08-20:
Can you try with
> LOG("SHARED_DRAM is at address: %lu\n", (long)(size_t)SHARED_DRAM);
> int weights_offset =
> ((long)(size_t)SHARED_DRAM_WEIGHTS - (long)(size_t)SHARED_DRAM) / sizeof(data_t);
> 
> int input_offset =
> ((long)(size_t)SHARED_DRAM_DATA - (long)(size_t)SHARED_DRAM) / sizeof(data_t);
> 
This should force the cast without errors...

RoxanaMir, 2017-08-22:
Thanks David, I did as you say and the error was gone. But after that I have a Syncheck error:
[SYNCHK-74] Recursive functions are not supported.

At  https://www.xilinx.com/support/answers/50495.html , I found the error but I don't know how solve it.

dgschwend, 2017-08-22:
Sorry, I don't know anything about this error, and I have never seen it before.
There should not be any recursive functions in the code. And others have successfully synthesized the code before.
What's the exact and complete error message?

RoxanaMir, 2017-08-23:
Dear David, 
Thanks for your answering. I was working with Vivado HLS 2015.4. I thought these errors may be for using this old version. Now I'm getting Vivado HlS 2017.1. I will synthesis the code with that and tell you the results.

Best Regards
Roxana

weiguochow, 2017-10-16:
Dear David,
Thanks very much for sharing the codes. I have some problems when running the testbench. Can you help me to solve this problem. Thank you!!!

FPGA: Computing .....@E Simulation failed: SIGSEGV.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source C:/Users/alex/Desktop/CNN/CNN/solution1/csim.tcl"
    invoked from within
"hls::main C:/Users/alex/Desktop/CNN/CNN/solution1/csim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"

dgschwend, 2017-10-17:
SIGSEGV would result when you access a memory area that you are not allowed to write to. Did you modify the code? What OS are you running? Which version of Vivado HLS?

weiguochow, 2017-10-17:
Dear David,
Thank you very much for your response.  I haven't modify the source code. and I run it on Windows10 and the Vivaodo HLS 2016.04
Thank you again. I am really appreciate your help.

It is always running until this line: 
**"CPU: Offload CONV Layer c10/p1:   8x8   x 736 > 512, CONV (1x1)/1        , IN @mem(13295616-13484032B), OUT @mem(13484032B), WEIGHTS @mem( 7096448- 8605824B) (split1) GLOBAL POOL "**

Best regards

dgschwend, 2017-10-17:
Hmm... I‘ve always used Linux instead of Windows, but it should not depend on the OS. I don‘t see what‘s going wrong here.
Unfortunately I’m quite busy now. Maybe I find some time to investigate next week.

weiguochow, 2017-10-17:
Thank you for your welcome answers! I will investigate more on the source code.

By the way, I am very interested in the field that implement CNN on FPGA. Can you introduce your current work? For convenience, do you have homepage or scientific research paper?

Best regards,
Weiguo Zhou

weiguochow, 2017-10-17:
Dear  @RoxanaMir ,

Have you run the testbench simulation successfully? Can you send me the whole HLS project? I wonder weather it had problems for operation systems, version of vivado hls or hardware cache memory of computer?

Best regards,
Weiguo Zhou

dgschwend, 2017-10-18:
I‘m currently working in industry and all related projects are under NDA. ;-)
Let‘s hope @RoxanaMir has some insights, otherwise I‘ll see if I can dig up my original Vivado project and compare.

RoxanaMir, 2017-10-21:
Hi Guys,
I was working with Vivado HLS 2015.4 at first and I had a lot of errors. Then I used Vivado HlS 2017.1 which I had non of the earlier errors but the project didn't run completely. In our country we use cracked soft wares and I think my errors is all for that. Then I wrote a code for a PE using C++ and run it in Vivado, but a also had a lot of errors. I really think using cracked soft ware is my problem.
Weiguochow use the testbench and source files table that David send above. It was a perfect guide for me. 
Best wishes for both of you.

retina007, 2017-10-24:
Dear @weiguochow     
I have the same problem with you (Simulation failed: SIGSEGV.) , do you solve the problem?  my vivado is 2016.4  with win7 system .     
my email is wdlong007@gmail.com  .   
waiting for your email.
Best regards

retina007, 2017-10-24:
@dgschwend   I have the same problem(Simulation failed: SIGSEGV.)    my vivado is 2016.4 with win7 system .        Can you share the HLS project to me.  
my email is wdlong007@gmail.com .
waiting for your email.
Best regards

Latootoo, 2017-12-14:
@retina007  I also get "Simulation failed: SIGSEGV" problem. And error turns out coming from the following lines:
fpga_top.cpp 
>> sprintf(fname, "blob/IN_%s.bin", layername);
>> sprintf(fname, "blob/OUT_%s.bin", layername);
>> sprintf(fname, "blob/WEIGHTS_%s.bin", layername);
 
After deleting the "blob/", everything works well.

Platform: Ubuntu 16.04, Vivado HLS 2017.01

You can choose Debug mode,  add breakpoints and track the error using "C Simulation -> Launch Debugger" step by step.

dgschwend, 2017-12-14:
Did you change the network layer names? You might need to increase the size of „fname“ from 30 characters to something bigger in that case.
Or are you maybe missing the „blob“ folder? Try creating a subfolder „blob“ in your project folder... or simply deactivate the „write DRAM to file“ part used for debugging (replace #if 1 with #if 0 in https://github.com/dgschwend/zynqnet/blob/21cf1cc61460794e2318ccb76aea2a5a7538de01/_HLS_CODE/fpga_top.cpp#L198)

dgschwend, 2017-12-14:
PS: According to issue #18, @retina007 solved his SIGSEGV problems by increasing SHARED DRAM size:
>> My earlier error is Simulation failed: SIGSEGV. In file cpu_top.cpp, I just change SHARED_DRAM size to more ; then the problem is solved.

Latootoo, 2017-12-14:
@dgschwend   Yes. Creating a "blob" subfolder  under "PROJECT/solution/build/" path before C-Simulation also works.

hex0102, 2018-02-09:
@dgschwend hi, david, how to increase the SHARED_DRAM SIZE, it seems to be precomputed by weightsize + datasize.  Do you mean to increase an arbitrary amount of storage?

dgschwend, 2018-02-10:
You can try to use for example 2*(weightsize + datasize) and see if that helps...

lanemo, 2018-05-25:
@RoxanaMir
Hi, sorry to bother you. I met the same "recursive error" with you when I tried building the HLS project. Did you solve the problem by using the latest version of Vivado? 
Btw, did you successfully build the whole zynqnet on FPGA at the end? 
I would appreciate it if you could kindly reply to me.

RoxanaMir, 2018-05-26:
Hi, I used the latest version of Vivado, but in our country some soft wares
are so expensive and banned, and those we can use are cracked, So I found
that the latest Vivado was cracked and it didn't work properly. So I didn't
continue with Vivado to the end. Sorry I can't help you with this problem.
I wish you the best with your project with Vivado.

YzwWh9327, 2018-06-11:
I met the same error.but now I solved it.You can expand the SHARED_DRAM to 4*(weightsize + datasize),and then you should modify DRAM_DEPTH = 23730304 in network.hpp.

faezeshadman, 2018-08-06:
Dear David,
Thanks very much for sharing the codes.
I want to try some other networks e.g. VGG or ResNet,so how can I generat .caffemodel file?

KrisYoung12, 2020-05-10:
@dgschwend Hi, David，I meet the "Error : File wegihts.bin could not been opened!" when I run the zynqnet on the petalinux, can you tell me how to resolve it?