dgschwend/zynqnet #18: about _FIRMWARE

retina007, 2017-10-25:
Dear @dgschwend 
I use the HLS generate the FPGA_TOP IPcore ,  and build the Vivado  Block Designer to generate the zynqnet_200M.bit file.    But _FIRMWARE confuse me.
1. if I use SDK ,then I copy all files(such as **fpga_top.cpp ... cpu_top.cpp** ,except document vivado_include and ZynqNet_Accelerator_HW_DEF) to SDK (does it right?) .  But  fpga_top.cpp  is HLS cpp style , can SDK compile it?
2. if  I  use **linux run on zynq xc7z045** , I  copy all the files to linux , then run "Makefile",  (does it right?)
3. document **vivado_include and ZynqNet_Accelerator_HW_DEF** are use for what?  
Thank you

dgschwend, 2017-10-26:
Glad to hear you managed to compile the HLS! What did you do to make it work?

Regarding firmware:
1. Yes, fpga_top.cpp can be compiled by a standard C++ compiler. The pragmas are simply ignored. That‘s one of the very cool features of HLS: The testbench can be executed all by itself.
2. Correct, but if you do not use SDK, you will need the „vivado_include“ folder (it contains some definitions required to compile the driver).
3. The HW_DEF is just an intermediate description of the IP and contains info about the addresses used to communicate with the FPGA. It‘s not actively used. For vivado_include, see 2.
Good luck!

retina007, 2017-10-26:
Dear @dgschwend 
My  earlier error is **Simulation failed: SIGSEGV**.    In file cpu_top.cpp, I just change SHARED_DRAM size to more ; then the problem is solved.
Today I export vivado hardware to SDK and  "build all" , but It report error below.(SDK 2016.4 run on win7 x64 )
[image]
[image]
I have some question : 
1. Is my SDK New Project setting right? (OS Platform: linux?)
2. I copy all the files(you can see from picture), but it report error (you also can see from picture), but etc/ap_private.h exist.
3. can you share the picture about the SDK **src file** in the SDK **project explorer**.

Thank you

dgschwend, 2017-10-27:
Ok, so the SIGSEGV error can be solved by increasing the shared DRAM size. Good to know! (@roxanamir)

I did not use the Xilinx SDK for building the software project, because we are running a customized Linux on the Zynq. However, I think you need to adjust your include directories. Normally, in Eclipse-based IDEs such as Xilinx SDK, you should be able to right-click > Project Settings in the Project Explorer, and somewhere there, you need to add the „src“ folder as include directory. That makes it possible to find the „etc/ap_private.h“ header. Alternatively, copy all the files under „src“ directly into the project root, without the extra „src“ directory level. That should work, too.

retina007, 2017-10-27:
Dear @dgschwend 
now every goes well.   IP core  and SDK works well.
finally, I want to know the zynq xc7z045 **configure**(in vivado block designer) , It is the .tcl file , you can generete it by following below picture. You can upload to github or email to me (wdlong007@gmail.com)
1. double click zynq7 block
[image]
2. click **presets** and choose **save configuration**
[image]
3. type the name (zynq) and choose the file path. and save the **.tcl file**
[image]

Good luck

weiguochow, 2017-10-30:
Dear David,

Thank you, I want to put live video or camera data into the Zynqnet module. How can I modify the project? Can you give me some advice.

Best regards

weiguochow, 2017-11-02:
Dear @retina007 ,
I am very glad to hear you have build vivado block design successfully, can you give me some tips on it?

Best regards

dgschwend, 2017-11-06:
@weiguochow: please open a separate issue for your questions

leexp0304, 2018-03-23:
@weiguochow  hello ,weiguochow,I come from Harbin Institute of Technology，CHINA. We also want to run cnn on FPGA ,my email is leexp1997@126.com. Can we contact by mail？thanks~

YzwWh9327, 2018-06-11:
@retina007 Hi!After generating fpga_top ipcore,how do you generate the zynqnet_200MHZ.bit file?

PXT846765038, 2019-03-25:
Hi David,
I have learned that your project can be run under the linux on board but my ZC706 doesn't have a linux on the SD card. I have found some tutorials about how to install a linux on ZC706 and how to start system from SD card but none of them worked. Can you give me some help? Thanks a lot.