dgschwend/zynqnet #19: about device and usage

LakyTT, 2017-11-06:
Hi ! @dgschwend  now I am learning CNN FPGA-based ，and so luckily to see your work here. I want to know whether it supports my FPGA. My FPGA is Xilinx Zynq UltraScale+MPSoC ZCU102.
if possible ，could you give me the steps about how to implement on FPGA？
I am a beginner ，so I am afraid I  need more details.

Best regards！！

dgschwend, 2017-11-06:
You need to compare the numbers from section 5.2.1 Resource Utilization with the number of LUTs, FFs, BRAMs and DSPs available on the XCZU9EG FPGA (the one on the ZCU102 eval board) for a rough guess. If it doesn‘t fit, you can still decrease the number of processing untis in the ZynqNet FPGA accelerator. However, the Ultrascale+ Platform may be better suited for the accelerator, so maybe you best make a test synthesis...
I‘m looking forward to your results, please keep this thread updated! :)

LakyTT, 2017-11-07:
thanks so much ！I will try to do it！

best regards！

LakyTT, 2017-11-10:
Hi ,dear david，now I am using vivado_hls，so what files should be test bench？and what files should be source in your folder ：_HLS_CODE ？

thanks ！

dgschwend, 2017-11-10:
You can see it e.g. in the screenshots in https://github.com/dgschwend/zynqnet/issues/11

LakyTT, 2017-11-17:
hi ！dear David .I am a bit confused now. can this network classify image in real-time?and how to  confirm the classification accuracy? how can we classify  images on the network via FPGA（not via GPU）?for example，inputing  an image via FPGA，can this network classify it?  forgive me for my stupid question.

best regards ！！

dgschwend, 2017-11-22:
Yes, ZynqNet can classify images in real-time (see the performance evaluations in the paper to see if it‘s fast enough for your real-time requirements). You can confirm the accuracy using the C++ golden model / HLS code. It gives bit-identical results to the FPGA implementation.
You need to run the accelerator + CNN on a Zynq SoC to classify images. You can feed the images from Linux or directly from within the FPGA (e.g. some custom IP).

LakyTT, 2017-11-23:
thanks so so so so much！！！now I  have got some good result.looking forward to next work！
when I end  my work ,I will tell  you  my final result.

best regards ！

closed 2017-12-12 by dgschwend

LakyTT, 2017-12-13:
hi！dear David：
I am using embedded Linux on ZCU102，and my result is as follows：

FPGA：computing........done.
FPGA：computing........done.
FPGA：computing........done.
CPU：Copy Results from FPGA DRAM（4096 Bytes）

Total run time：2237400ms

Result(top-5)：
=============
89.05%：class 207 （output 21.10）
4.09%：  class 208 （output 18.02）
3.37%：  class 219 （output 17.83）
2.13%：  class 852 （output 17.37）
0.36%：  class 257 （output 15.58）

TestBench Result：FAILURE
Actual : 89.05,Expected: 88.38

so there are three questions：
1.Is my result  right? why my actual result >expected result？
2.class 207、class 208 represent  what？and why total run time  is so long？image I used is your indata.bin
3.how to load my cnn.bit to my board（ZCU102）？and how to know whether it runs on my FPGA or my  PS of my board？

looking forward your reply   
                                                                                                     best regards！！thank you very much！

dgschwend, 2017-12-13:
How did you run this test?

LakyTT, 2017-12-13:
first execute the Makefile，then execute test.
according to tips："Usage: ./test  CPU|FPGA input_file [-quiet]"
Type "Usage:./test CPU input_file [-quiet]" and get this result.

LakyTT, 2017-12-15:
could you give me  some suggestion about my questions please？
maybe my questions are stupid（I am a  rookie in FPGA），forgive me..

best regards！ thank you！

dgschwend, 2017-12-15:
You are running the CPU Simulation. You need to build the project using Vivado HLS first, then generate the Bitstream using Vivado, then generate the Zynq Software using Xilinx SDK. Then you can run on FPGA.
You can find the association between class IDs and class names for ImageNet for example here:
https://gist.github.com/yrevar/942d3a0ac09ec9e5eb3a