dgschwend/zynqnet #22: about how to open the project

wswsamao, 2017-12-11:
Hi David. 
       Thanks for your sharing, I am a beginner. I am very interested in your design and I have download the code, but I found that I can not open this project by Vivado, could you please share the project file？
    Thank you again
    Best wishes
    Shuo Wang

dgschwend, 2017-12-11:
Sorry,cannot share the project file due to licensing concerns. See this screenshot https://cloud.githubusercontent.com/assets/23473441/24205561/bad92364-0f41-11e7-8dbc-5f3dd832a721.png as an example configuration, and be sure to check the other issues here (also the closed ones).

wswsamao, 2017-12-12:
thanks again, I will try it

jc5706, 2018-03-01:
That helps to know what is application and what is test bench