dgschwend/zynqnet #24: Can zynqnet run on xilinx zcu102?

lishen565, 2017-12-18:
Hi all, our lab has only a soc xilinx zcu102 and i'm a freshman of FPGA developing. I want to verify zynqnet on zcu102 so i have to transplant it from 7045 to zcu102. Are there any docs or other projects to help me run zynqnet on zcu102? Thank you all.

lishen565, 2017-12-18:
link of xilinx zcu102 :https://china.xilinx.com/products/boards-and-kits/ek-u1-zcu102-g.html#trainingSupport

lishen565, 2017-12-18:
link of zcu102 https://www.xilinx.com/products/boards-and-kits/ek-u1-zcu102-g.html#trainingSupport

dgschwend, 2017-12-18:
The Xilinx Ultrascale ZU9 on your board is significantly bigger than the FPGA fabric on the Zynq 7045 I used. There should be no problem using this board. As for docs: follow the infos in the project report and the Xilinx HLS Guide / „ultrafast design methodology“ guides + videos to get started with HLS. Let me know if you have any specific problems and questions.

lishen565, 2017-12-20:
Thank you for your answer. Could you tell me the specific type of xilinx Soc used for Zynqnet? And our lab will buy it to start work on new board.

dgschwend, 2017-12-20:
The details are all in the project report. I‘d recommend an evaluation kit with some kind of video output in case you want to do a demo.

leexp0304, 2018-03-23:
@lishen565 hello ,lishen565,I come from Harbin Institute of Technology，CHINA. We also want to run zynqnet on zcu102,my email is leexp1997@126.com.  Can we contact  by mail？thanks~

wxbbuaa2011, 2018-06-11:
@leexp0304 the same to you, my mail is wxbbuaa2011@163.com

Sunyouteng, 2018-12-19:
want to run it on zcu102 too. sunyouteng@163.com