dgschwend/zynqnet #25: input image for simulation

gueyezizaaa, 2017-12-19:
hello I'm currently working on your project and I want to know which kind image you used for your siumulation 
Also, I'm actually trying to convert jpg image with the python code provided but that dosen't work 
this is the line code "  convert_image("‪f:\telecharge.jpg",225,225,"‪C:\Users\Asus\Desktop")  " 
is there an error.
thanks

dgschwend, 2017-12-20:
Image should be RGB. JPG converted with the Python code should work fine. What‘s the exact error?

gueyezizaaa, 2017-12-20:
this is the msg
[image]

dgschwend, 2017-12-30:
That error looks strange, as if there went something wrong during copy-pasting the code. Are you using Python3, and the code from https://github.com/dgschwend/zynqnet/issues/4#issuecomment-283332644?