dgschwend/zynqnet #26: "OPMODE Input Warning"  Discussion

Latootoo, 2017-12-22:
I have tried to fixed the "OPMODE Input Warning" . And this warning sames coming for "fpga_top_fadd_32ng8j.v" alone, which adds two floating data (32bit) using DSP48E.  Other files including "fixed-point data ADD/MUL" and "float-point data MUL/CMP" do not  induce this warning. 
    This conclusion has verified after varying the _data type_ and _function_ of the Xilinx Tutorial "ug871-design-files/Using_IP_with_Zynq/lab1/hls_macc" .  
`void hls_macc(**int** a, **int** b, **int** *accum, bool accum_clr)
{
   static acc_reg = 0;
   if (accum_clr)
      acc_reg = 0;
   **acc_reg += a * b;**
   *accum = acc_reg;
}`   

   It sames that using "fixed-point ADD" to replace "float-point ADD" is the only solution to avoid this warning.  But it's really inelegant. And I still don't know the reason. Could any one give me more reminder?

dgschwend, 2017-12-22:
So this is still not fixed in the Xilinx libraries...
It‘s just a warning, you can ignore it and the generated HDL code / FPGA logic still works. However, the warning slows down simulation so much that proper HDL simulation is not possible.
Someone would need to file a bug report to Xilinx.

dgschwend, 2017-12-22:
But anyway, using fixed-point instead of float would be a great idea for ZynqNet!