dgschwend/zynqnet #27: DEBUGGING the Code

wasim6691, 2017-12-23:
Hi David

Hope You are fine. I have to ask You that can we compile and Debug  Your code on a c++ compiler besides VIVADO HLS. I really want to debug your code to understand it fully. Thanks. waiting for Your reply

Khan
Master thesis student at Bosch, Reningen.

Latootoo, 2017-12-24:
Vivado HLS can easily debug the code step by step. Other c++ compiler maybe unable to handle #pragma directives.

wasim6691, 2017-12-24:
Which Fpga should I select in the vivado Open Project. I am encountering an
error when I run the project. I send u the exact error in some time.

dgschwend, 2017-12-26:
Most FPGAs should work. Original work is on XCZ7045. See the project report for details.

closed 2017-12-30 by dgschwend