dgschwend/zynqnet #29: zynqnet on xilinx XC7Z020

ortalik, 2017-12-30:
Hello David,

I am interested in testing your network on the XC7Z020 SoC which is smaller than the XC7Z045 you were using. I can see that the main issue is with the bram utilization.
Do you have an idea how this could work? Are there any changes we could try? Or do we need to completely redesign?

Thank you!

mslavescu, 2018-03-15:
I'm also interested in a variant of this network for PYNQ-Z1 board, that uses ZYNQ XC7Z020-1CLG400C:
https://store.digilentinc.com/pynq-z1-python-productivity-for-zynq/

ysdagar, 2018-03-17:
I'm also interested in a variant of this network for Xilinx-ZC702 Evaluation Kit that uses ZYNQ XC7Z020 Soc.

JohnnyOpcode, 2018-03-17:
I'm fascinated by the people who want this on a Z7020. I guess they can't comprehend the FPGA resources required for this magnificent piece of engineering. I guess they also can't try and optimize the code further and expect someone (the author) to do it for them. My suggestion would be to study this work of art, learn and stand on the shoulders of greats like dgschwend. I dare you all to do better.

ysdagar, 2018-03-18:
Thank you Johnyopcode, for your encouraging words & yes challenge taken! :) :)