dgschwend/zynqnet #30: Problem in exporting RTL in VHLS.

lishen565, 2018-01-02:
Hi, I've passed the C simulation and C synthesis, but it took too long time(about five days) to finish the C/RTL cosimulation. So I dropped the cosimulation and wanted to  export the RTL , but now I get stuck at this step. The detail info VHLS told me is below, could you please help me out，thank you？

 Starting export RTL ...
D:/Xilinx_Vivado_2016.4/Vivado_HLS/2016.4/bin/vivado_hls.bat D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1/export.tcl
INFO: [HLS 200-10] Running 'D:/Xilinx_Vivado_2016.4/Vivado_HLS/2016.4/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'lishen' on host 'amax-pc' (Windows NT_amd64 version 6.1) on Mon Jan 02 20:21:39 +0800 2023
INFO: [HLS 200-10] In directory 'D:/lishen/lijun/zynqnet-master'
INFO: [HLS 200-10] Opening project 'D:/lishen/lijun/zynqnet-master/zyncnet_ls1'.
INFO: [HLS 200-10] Opening solution 'D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-10] Setting target device to 'xc7z045ffg900-2'
INFO: [IMPL 213-8] Exporting RTL as an IP in IP-XACT.

****** Vivado v2016.4 (64-bit)
  **** SW Build 1756540 on Mon Jan 23 19:11:23 MST 2017
  **** IP Build 1755317 on Mon Jan 23 20:30:07 MST 2017
    ** Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.

source run_ippack.tcl -notrace
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository 'D:/Xilinx_Vivado_2016.4/Vivado/2016.4/data/ip'.
WARNING: [IP_Flow 19-4832] The IP name 'fpga_top_ap_fadd_2_full_dsp_32' you have specified is long. The Windows operating system has path length limitations. It is recommended you use shorter names to reduce the likelihood of issues.
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'fpga_top_ap_fadd_2_full_dsp_32'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'fpga_top_ap_fadd_2_full_dsp_32'...
WARNING: [IP_Flow 19-4832] The IP name 'fpga_top_ap_fcmp_0_no_dsp_32' you have specified is long. The Windows operating system has path length limitations. It is recommended you use shorter names to reduce the likelihood of issues.
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'fpga_top_ap_fcmp_0_no_dsp_32'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'fpga_top_ap_fcmp_0_no_dsp_32'...
WARNING: [IP_Flow 19-4832] The IP name 'fpga_top_ap_fmul_1_max_dsp_32' you have specified is long. The Windows operating system has path length limitations. It is recommended you use shorter names to reduce the likelihood of issues.
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'fpga_top_ap_fmul_1_max_dsp_32'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'fpga_top_ap_fmul_1_max_dsp_32'...
bad lexical cast: source type value could not be interpreted as target
    while executing
"rdi::set_property core_revision 2301022021 {component component_1}"
    invoked from within
"set_property core_revision $Revision $core"
    (file "run_ippack.tcl" line 1042)
INFO: [Common 17-206] Exiting Vivado at Mon Jan 02 20:22:12 2023...
ERROR: [IMPL 213-28] Failed to generate IP.
command 'ap_source' returned error code
    while executing
"source D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1/export.tcl"
    invoked from within
"hls::main D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1/export.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished export RTL.

Starting export RTL ...
D:/Xilinx_Vivado_2016.4/Vivado_HLS/2016.4/bin/vivado_hls.bat D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1/export.tcl
INFO: [HLS 200-10] Running 'D:/Xilinx_Vivado_2016.4/Vivado_HLS/2016.4/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'lishen' on host 'amax-pc' (Windows NT_amd64 version 6.1) on Mon Jan 02 20:31:18 +0800 2023
INFO: [HLS 200-10] In directory 'D:/lishen/lijun/zynqnet-master'
INFO: [HLS 200-10] Opening project 'D:/lishen/lijun/zynqnet-master/zyncnet_ls1'.
INFO: [HLS 200-10] Opening solution 'D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-10] Setting target device to 'xc7z045ffg900-2'
INFO: [IMPL 213-8] Exporting RTL as an IP in IP-XACT.

****** Vivado v2016.4 (64-bit)
  **** SW Build 1756540 on Mon Jan 23 19:11:23 MST 2017
  **** IP Build 1755317 on Mon Jan 23 20:30:07 MST 2017
    ** Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.

source run_ippack.tcl -notrace
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository 'D:/Xilinx_Vivado_2016.4/Vivado/2016.4/data/ip'.
WARNING: [IP_Flow 19-4832] The IP name 'fpga_top_ap_fadd_2_full_dsp_32' you have specified is long. The Windows operating system has path length limitations. It is recommended you use shorter names to reduce the likelihood of issues.
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'fpga_top_ap_fadd_2_full_dsp_32'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'fpga_top_ap_fadd_2_full_dsp_32'...
WARNING: [IP_Flow 19-4832] The IP name 'fpga_top_ap_fcmp_0_no_dsp_32' you have specified is long. The Windows operating system has path length limitations. It is recommended you use shorter names to reduce the likelihood of issues.
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'fpga_top_ap_fcmp_0_no_dsp_32'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'fpga_top_ap_fcmp_0_no_dsp_32'...
WARNING: [IP_Flow 19-4832] The IP name 'fpga_top_ap_fmul_1_max_dsp_32' you have specified is long. The Windows operating system has path length limitations. It is recommended you use shorter names to reduce the likelihood of issues.
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'fpga_top_ap_fmul_1_max_dsp_32'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'fpga_top_ap_fmul_1_max_dsp_32'...
bad lexical cast: source type value could not be interpreted as target
    while executing
"rdi::set_property core_revision 2301022031 {component component_1}"
    invoked from within
"set_property core_revision $Revision $core"
    (file "run_ippack.tcl" line 1042)
INFO: [Common 17-206] Exiting Vivado at Mon Jan 02 20:31:32 2023...
ERROR: [IMPL 213-28] Failed to generate IP.
command 'ap_source' returned error code
    while executing
"source D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1/export.tcl"
    invoked from within
"hls::main D:/lishen/lijun/zynqnet-master/zyncnet_ls1/solution1/export.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished export RTL.

lishen565, 2018-03-07:
[image]

lishen565, 2018-03-07:
Have you met this error during "export RTL"?
[image]

dgschwend, 2018-03-07:
No, I’ve never seen this problem before. Looks like a bug in the IP packaging code. Have you tried packing the IP in a different format (Verilog/VHDL/all other settings)? Can you try a different version of VHLS? Maybe like that you can find a way around that problem...

lishen565, 2018-03-08:
Thanks, i'll try the ways you advised.

PXT846765038, 2018-05-05:
Hi, @lishen565 I met the same problem after C synthesis and C simulation. Have you solve it? Looking forward your help. Thank you.

LIUQyou, 2022-01-01:
Hi, did you solve the problem, I encounter the same problem

eejlny, 2022-01-01:
Sorry, I have the same problem and I can see you got the same problem 2 hours ago. Something to do with the new year ? I never had this problem with the same project yesterday 31st Dec but today 1st Jan is not working with this error.

LIUQyou, 2022-01-01:
Me too, it is very weird. I try to export the RTL into IP with the same code, but I encountered this problem. Tomorrow I will try it on a new version of HLS. But for now, I have no idea about the solution. sorry

beatsnbytes, 2022-01-01:
Same problem here!

jtani78, 2022-01-01:
I have the same problem.
Workaround: After setting the system time of windows to year 2021, it is working.
But I really hope, that there is a better solution for this problem.

Any ideas?

bwajster97, 2022-01-02:
Got the same error. Anyone able sort this out yet ?(apart from changing the system time)

jtani78, 2022-01-02:
What I found in the solution1/impl/ip/vivado.log:
`source run_ippack.tcl -notrace
bad lexical cast: source type value could not be interpreted as target
    while executing
"rdi::set_property core_revision 2201021649 {component component_1}"
    invoked from within
"set_property core_revision $Revision $core"
    (file "run_ippack.tcl" line 963)
INFO: [Common 17-206] Exiting Vivado at Sun Jan 02 16:49:39 2022...
`
Line 963 of solution1/impl/ip/run_ippack.tcl:
`set_property core_revision $Revision $core`

I made a diff between the working and not working version --> it is just the datecode
Do anyone know, where the file run_ippack.tcl comes from? Is there a template or something else?
If yes, we could change it perhaps. If it is hardcoded in vivado.exe, we have a big problem.

jtani78, 2022-01-02:
I think, I found the reason of the problem:

All datetimes from 2021 are working. 2022 is not working.

Here is some math:
02.01.202**1** 16:49 results in core_revision **2101021649** --> equals to hex number **7**D3B0BD1
--> binary **0**1111101001110110000101111010001

02.01.202**2** 16:49 results in core_revision **2201021649** --> equals to hex number **8**330ECD1
--> binary **1**0000011001100001110110011010001

The very left bit of the number 8330ECD1 is set to '1' --> if this variable is a 32bit signed integer --> the value gets negative (because of Two's complement). I would bet, that this is the problem.

Solution: We should change the line 963 to something else. But this file is autogenerated.
Any ideas how to change this?

eejlny, 2022-01-02:
Congrats, great detective work finding the reason .... Not sure about the file though,

jtani78, 2022-01-02:
Thank you :-)
But now is the big question, how to change the behavior . Do you want to look at the xilinx installation folder? Perhaps you will find something in the TCL code.
Would be really cool.

I already searched for a workaround. Without any success until now. There are so much files in the HLS installation folder... :-(

Further idea: Perhaps we should try to export the IP manually (via a script)?

brostvincent, 2022-01-03:
Hello, same problem. The solution for me today :  back to the futur...
xilinx wish you a happy new year :)

tux2603, 2022-01-04:
I just reproduced the issue in extremely simple MWE in Vivado 2019.1. That seems to confirm that it's upstream. Has anyone reported this to xilinx yet?

brostvincent, 2022-01-04:
The first solution to work around the problem. Roll back the date: not  easy. the second, set configuration of IP with number version x.y.z

beatsnbytes, 2022-01-04:
And how do you perform the second solution @brostvincent without GUI?

brostvincent, 2022-01-04:
In HLS GUI
Export -> configuration -> Version -> 0.0.0

jtani78, 2022-01-04:
Thank you brostvincent!
It is working.

Happy new year xilinx :angry:

eejlny, 2022-01-04:
No gui for me either. Just Makefiles runing sds++ :(

beatsnbytes, 2022-01-04:
https://support.xilinx.com/s/question/0D52E00006uxYjmSAE/bug-report-vivado-20212-revision-number-overflow-after-new-year?language=en_US

bwajster97, 2022-01-05:
> In HLS GUI Export -> configuration -> Version -> 0.0.0

Does changing the version (under export ->configurations) from 1.0 to 0.0 solve the problem ?

bwajster97, 2022-01-05:
> The first solution to work around the problem. Roll back the date: not easy. the second, set configuration of IP with number version x.y.z

Kindly pls explain what you mean by number version. (I'm using version 2019.1)

ysdagar, 2022-01-05:
Hi,
I have been observing a lot of discussion here.  it's been looked upon by Xilinx. Once i get an update I'll post here.

Cheers,
Yashwant

tangshiren1993, 2022-01-06:
Export -> configuration -> Version -> 1.0.1

bwajster97, 2022-01-06:
> Export -> configuration -> Version -> 1.0.1

Why do we set the version to 1.0.1 ?

jtani78, 2022-01-06:
If you do not write something to the field "version", the version will be calculated by VIVADO. This will result in the error described above.
If you write something into the field "version", VIVADO will not try to calculate the version and we will get no error. GREAT!

It is your decision, what version you want... For the first export, you can use e.g. 1.0.0 for the second 1.0.1 for the third 1.0.2 and so on.
For major changes you could change the first number e.g. 2.0.0

bwajster97, 2022-01-06:
Thank You so much! @jtani78

mgh-prog, 2022-01-11:
thank you man! @brostvincent

antonpapazyan, 2022-01-14:
Look at this post, here is the temporary solution to the issue.
https://support.xilinx.com/s/question/0D52E00006vDlvJSAS/export-ip-invalid-argument-revision-number-overflow-issue?language=en_US

arpanpoudel, 2022-03-15:
@jtani78 @brostvincent thank you so much, saved me

yhkwon6658, 2023-07-27:
@brostvincent Thank you bro!!

shawaizhaider, 2024-07-03:
@brostvincent and  @jtani78  Thank you so much, guys!!

phamtiep, 2024-12-18:
tks u so much guys its work ....
i have deadline around 2hour ....