dgschwend/zynqnet #31: Image Pixel Values Range

wasim6691, 2018-01-09:
Hi David

I want to ask You that If i convert the hexadecimal values from the "indata.bin" file. What range of the values should I expect. At the moment I am getting the range of the pixels values in negative and positive both. Is there any defined range of pixel values e.g. (0, 255) or (-255, 255) etc. Hoping for an early response. Thanks

Best Regards
Khan

dgschwend, 2018-01-09:
Good question! As you can see in the conversion script https://github.com/dgschwend/zynqnet/blob/master/tools/convert_jpeg_to_indata_bin.py, Zynqnet expects inputs between 0-255, but with the ImageNet mean vales [104, 117, 123] subtracted.

dgschwend, 2018-01-09:
I.e. the floating-point range will be approximately -123 to +151, depending on the channel (R/G/B)

wasim6691, 2018-01-09:
I am reading the indata.bin file and making a 3d array for storing the
pixel values. The array pattern is [channels][height][Width]. Is that
pattern ok ? But I have not subtracted any mean value from the values which
I m reading from the binary file ?

dgschwend, 2018-01-09:
CHW order is correct. Beware of BGR color channels.
See the linked script for the conversion from JPG -> indata.bin, it says all about the conversion.
Why are you trying to interpret the binary file, by the way?

wasim6691, 2018-01-09:
The binary file indata.bin is of specific image right?. So I am reading the
values from the binary file and trying to implement the first layer my
self. Can we use the binary file as image pixel file. The values inside are
in hexa and I m interpreting 4 hex values as 1 floating point value ? Is
that right or wrong ?

wasim6691, 2018-01-09:
Hi David 

Please Answer the Above question. 

One More question is that why are you subtracting the mean values for the ImageNet image Set. Can You please explain the reason behind tat. Thanks for your time. Keenly looking forward for your reply.

dgschwend, 2018-01-10:
Your assumptions are correct.

The network has been trained with mean-subtracted images, therefore needs mean-subtraction also during inference. There‘s some debate whether this is necessary in the Deep Learning community, but it‘s standard in the DIGITS framework I trained the network with.

wasim6691, 2018-01-10:
HI David

I am.getting the floating point values in the range of -123 to 151 as you
have mentioned before. What do you think it would be ok to use these as
pixels values for deep neural networks like zynqnet. I am using these and
multiplying with the weights values to get the next layer values.? Thanks