dgschwend/zynqnet #32: Problem in generating bitstream with Vivado SDSOC

HeroGian, 2018-01-16:
Dear David,

I'm a self driving car researcher of the university of Modena. I'm very interested in your work and I want to test the ZynqNet on Zynq Ultrascale.
I have tested the net with Vivado HLS simulation and it works perfectly, but when I try to synthetize fpga_top in hardware with Vivado SDSOC it returns me some DMAnalysis errors, in particular:

`ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1305] Trying to overwrite port 'layer_i' on CF block 'hwblk_fpga_top'
ERROR: [DMAnalysis 83-1305] portType = stream, expected stream
ERROR: [DMAnalysis 83-1305] direction = in, expected in
ERROR: [DMAnalysis 83-1305] mode = slave, expected master
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1369] NULL destination port 0x3bf4a90
ERROR: [DMAnalysis 83-1378] NULL  -->  s_axi_axilite offset:0x10
ERROR: [DMAnalysis 83-1332] CF data model: port map [2] error.  Block: hwblk_fpga_top, Comp: fpga_top_1
ERROR: [DMAnalysis 83-4447] Failed creating data motion network hardware!`

I have tried to remove the slave HLS pragmas under the fpga_top declaration in fpga_top.cpp, but I get this port mapping error:

`ERROR: [HSL2XD 83-101] The HLS function 'fpga_top' has an invalid port mapping.  When any argument is mapped onto an axilite interface, the return value must be mapped to the same interface, e.g., with #pragma HLS interface s_axilite port=return bundle=s_axi_AXILiteS
`
Could yout give me some helps with these errors? I'm new in FPGA programming

Best Regards,
Gianluca

dgschwend, 2018-01-16:
Hi Gianluca

Thanks for your interest. I did not use SDSoC for the bitstream compilation, so unfortunately I don‘t have any ideas regarding your errors. I would suggest you read the HLS guides and SDSoC guides, especially the sections related to interfaces. You could also post on the Xilinx forums.

Good luck! David

HeroGian, 2018-01-17:
Thank you for the answer David,  I'll read the Vivado HLS and SDSOC manuals and I hope I can solve the problem

bkzshabbaz, 2018-02-08:
@HeroGian were you able to resolve your issue?  I'm also trying to compile this project with SDSoC and am experiencing the same errors.

HeroGian, 2018-02-09:
Hi, I managed to sinthetize ZynqNet with SDSoC v2017.1, I have made changes in some pragmas.
Try to remove these pragmas in fpga_top.cpp:
`#pragma HLS INTERFACE m_axi depth = DRAM_DEPTH port = SHARED_DRAM offset = slave bundle = memorybus register
#pragma HLS INTERFACE s_axilite port = layer bundle = axilite  register
#pragma HLS INTERFACE s_axilite port = num_weights bundle = axilite  register
#pragma HLS INTERFACE s_axilite port = weights_offset bundle = axilite  register
#pragma HLS INTERFACE s_axilite port = input_offset bundle = axilite  register
#pragma HLS INTERFACE s_axilite port = return bundle = axilite  register `

and replace with these:

`#pragma SDS data mem_attribute(SHARED_DRAM:PHYSICAL_CONTIGUOUS)
 #pragma SDS data zero_copy(SHARED_DRAM[0:DRAM_DEPTH])`

jc5706, 2018-03-01:
HeroGian, as far as I know, SDSoC is meant to translate C functions into PL blocks, while the original intention is to run that software on the embedded ARM cores. I would use the SDK instead.

wxbbuaa2011, 2018-06-11:
@HeroGian can you send me a project of SDSoC？my mail is wxbbuaa2011@163.com,Thank you very much.