dgschwend/zynqnet #35: HLS Synthesis Warnings

haroonrl, 2018-02-01:
Hi , 
i am unable to synthesis the accelerator , synthesis results in following errors and warnings:

ERROR: [HLS 200-70] Synthesizability check failed.
command 'ap_source' returned error code
    while executing
"source /opt/Xilinx/Vivado/2017.4/bin/fpga/solution1/csynth.tcl"
    invoked from within
"hls::main /opt/Xilinx/Vivado/2017.4/bin/fpga/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"  fpga:solution1  Feb 1, 2018 1:28:37 AM

ERROR: [SYNCHK 200-61] ../../../../../home/harry/Downloads/zynqnet-master/_HLS_CODE/memory_controller.cpp:123: 
unsupported memory access on variable 'SHARED_DRAM' which is (or contains) an array with unknown size at compile time.  fpga:solution1  Feb 1, 2018 1:28:37 AM

Warnings are:
@W[GUI]:"port" is required; "bundle" "input"  invalid; "" ""  invalid

please help me to solve this .

leexp0304, 2018-03-25:
@haroonrl ,你好，可以试试吧shared_bram改成原来的两倍大小再看看，我改完以后已经综合成功。

Xuexiabc, 2023-02-08:
> 

我也遇到了相同的问题，请问shared_bram在哪里修改