dgschwend/zynqnet #36: Convert jpeg to binary

suzinee, 2018-02-21:
Hi :)
I've got some issues. Would you check it up for me?
I found a different result from the example you uploaded(indata.bin) while trying to convert puppy-500x350.jpg to test.bin.
After converting the puppy-500x350.jpg, I compared test.bin with indata.bin except for noise padding parts and I used half-fill, half-crop by uncomment your code. Unfortunately, they don't have the same value exactly.
Here are my results.
[Note. Black pixels mean different values and White pixels are equivalent values]
[image]

I worried that I would get wrong binary files in this way when I converted other jpg/jpeg files.
Actually, I checked the low accuracy of test.bin I converted around 75% when trying in CPU simulation, which means whereas the original file (indata.bin) shows the accuracy around 88% but I couldn't get.

Below the image shows the result of CPU simulation.
[Note. Left one uses indata.bin Right one uses test.bin]
[image]

Could you tell me why didn't I get the accuracy?
and happy new year!

dgschwend, 2018-02-27:
Hi

I‘m not sure why there‘s a binary difference. Can you check how much the individual pixel values differ? E.g. by how many percent you‘re off from the original image, for some test pixels? It‘s normal that floating-point computations give slightly different results on different machines, but it should be only a very small difference.

Then, what you‘re referring to as 88% vs. 75%„accuracy“ has nothing to do with actual CNN accuracy. It‘s just the Softmax probability that the network assigns to the „dog“ class. As long as your network also assigns the highest probability to class 207 („dog“), all is fine.
If you want to measure the CNN’s accuracy, you have to convert and infer the entire ImageNet test set and measure the percentage of correct guesses.

Regards
David

suzinee, 2018-02-28:
Thank you for your reply:)

OHhh there was the confusing word "ACCURACY" I wrote. As you said, it's just the probabilities calculated from the Softmax function. What I wanted to tell you was that I got the lower probabilities when I used test.bin than indata.bin.

Anyway, since the result of top-1 class is equivalent to "207", I don't care any problem to use the code I removed comments when I use new images converted by the code I uploaded, do I?

Regards
Sujin

dgschwend, 2018-02-28:
Yes, as long as the top-1 class is the same and has a clearly higher Softmax class probability than the rest of the classes, you don‘t need to care. 

The network was trained with „half-crop-half-fill“ images, so I assume this method is also reasonable for inference. You would have to try and measure the impact on accuracy...

suzinee, 2018-02-28:
It's very nice of you to say so.
Thank you, David!