dgschwend/zynqnet #41: Problems about C cimulation and ZYNQ Board

JackChenofNCP, 2018-03-31:
Hi, David. I am interested in you zynqnet. It's really a great work. Thanks for sharing. I learnt a lot from you report and code. So I am trying to test you CNN on my own ZedBoard these days and have met 2 problems.

1. I have imported you HLS code files into my HLS project correctly according to one of your previous reply to @wswsamao and synthesised successfully. But I met these errors while running C simulatio:
---------------------------------------------------------------
FPGA: Computing .....@E Simulation failed: SIGSEGV.
@E [SIM-1] CSim failed with errors.
4
    while executing
"source D:/Users/Administrator/zynqnet/solution1/csim.tcl"
    invoked from within
"hls::main D:/Users/Administrator/zynqnet/solution1/csim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
----------------------------------------------------------------
I am not quite familiar with HLS. Could you please help me figuring out why this occured? Thank you.

2. I checked the "Evaluation and Results" in your report and found that the resource ultilization exceeds the range of my ZedBoard. You used 996 BLOCK RAMs for you CNN design but I only get 500 on my board. 
So it seems I have to make some simplification to your zynqnet in order to test it on my board. And I know of course that will damage the performance of your CNN but I really want to make it work on my board. So could you give me some advice on how to shrinking down the size of your zynqnet without causing too much damage to the performance? Thanks. Looking forward to your reply.

dgschwend, 2018-04-01:
1. the SIGSEGV error can be fixed by increasing the size of SHARED_DRAM (e.g. by a factor of 2), as others here have found out
2. if you want to shrink the network, you need to have an entire CNN training toolchain ready (Caffe, GPU Server, ImageNet database). How to shrink a CNN with minimum impact on accuracy is an open question in science and currently boils down to trial & error. You certainly need to identify the layer which are currently the largest (thus defining the required BRAMs) and start trimming those first.
You might also try to adjust the FPGA architecture instead, and reduce the number of memories concurrently used there. This is not an easy task either!
Good luck!

JackChenofNCP, 2018-04-02:
Thanks for your solution. Increasing the size of SHARED_DRAM is right. I am working on the simplification now and may need you help again if new questions accures.

JackChenofNCP, 2018-04-02:
Thanks for your solution. Increasing the size of SHARED_DRAM is right. I am working on the simplification now and may need your help again if new trouble occures.

guwenning, 2018-05-15:
@JackChenofNCP  I have a question. I would like to ask how to increase the size of SHARED_DRAM. How to achieve specific. THANKS

PSlearner, 2018-10-23:
> @JackChenofNCP I have a question. I would like to ask how to increase the size of SHARED_DRAM. How to achieve specific. THANKS

In the cpu_top.cpp, see the line 199, SHARED_DRAM = (char *)malloc(total_size);
replace the line with SHARED_DRAM = (char *)malloc(2*total_size);
Then it worked.

Sunyouteng, 2018-12-24:
@why did we have to double it? can we get the exact space for SHARED_DRAM ?

Sunyouteng, 2018-12-24:
@PSlearner why did we have to double it? can we get the exact space for SHARED_DRAM ?