dgschwend/zynqnet #42: How to debug/run the C project  “zynqnet_sdk” in Xilinx SDK？

lishen565, 2018-04-07:
Hi, I've finished all the steps in HLS and Vivado. But when i launch the xilinx SDK and start to debug the C project "zynqnet_sdk" which has a main() in cpu_top.cpp, the xilinx SDK tells me "Launch Failed. Bianry not found".  Did you meet this problem before？My project explorer info is as follow.

lishen565, 2018-04-07:
[image]

lishen565, 2018-04-07:
when i start to "debug as" -> "4 launch on hardware(GDB)", SDK info is as below.
[image]

dgschwend, 2018-04-08:
Sorry but I did not use Xilinx SDK, this seems to be a specific problem related to the IDE...

lishen565, 2018-04-09:
Thanks. I've solved this problem in Xilinx SDK. Now I boot my ZC706 board  based on the petalinux(http://www.wiki.xilinx.com/Zynq2017.4+Release). I put the image of petalinux in SD card and debug the zynqnet code in "_FIRMWARE" dir. But the petalinux image has no libstdc++.so.6 and I can't continue the debug under Xilinx SDK. 
[image]

lishen565, 2018-04-10:
I’ve solved the problem by configuring the linker with "-static".

wxbbuaa2011, 2018-06-11:
@lishen565 can you send me a project of SDK? My mail is wxbbuaa2011@163.com.Thank you very much.

KrisYoung12, 2020-05-07:
@lishen565 Hi, I've meet the same problem. Can you tell me how to solve the problem by configuring the linker with "-static"? I don't know how to configure the linker with "-static".

ihaterecursion, 2021-01-07:
> Hi, I've finished all the steps in HLS and Vivado. But when i launch the xilinx SDK and start to debug the C project "zynqnet_sdk" which has a main() in cpu_top.cpp, the xilinx SDK tells me "Launch Failed. Bianry not found". Did you meet this problem before？My project explorer info is as follow.

i meet some problems when i build my project, could you plz give me some advice?

ALEX5874, 2021-03-30:
> 谢谢。我已经在西林克斯SDK解决了这个问题。现在我启动我的ZC706板基于花瓣[（http://www.wiki.xilinx.com/Zynq2017.4+Release）。](http://www.wiki.xilinx.com/Zynq2017.4+Release)我把花瓣的图像放在SD卡中，并在"_FIRMWARE"dir中调试zynqnet代码。但花瓣图像没有 libstdc +.so.6， 我不能继续在 Xilinx Sdk 下进行试试。
> [image]

hello, I've met the same problem too. Can you tell me how to solve the problem by configuring the linker with "-static"? I don't know how to configure the linker with "-static".

ALEX5874, 2021-03-30:
> @lishen565 Hi, I've meet the same problem. Can you tell me how to solve the problem by configuring the linker with "-static"? I don't know how to configure the linker with "-static".

hi,do you know how to solve the question now?

ALEX5874, 2021-03-30:
> Thanks. I've solved this problem in Xilinx SDK. Now I boot my ZC706 board based on the petalinux(http://www.wiki.xilinx.com/Zynq2017.4+Release). I put the image of petalinux in SD card and debug the [...]

HI, I also use the petalinux to make linux available on the board, I want to know do you use some other things or just what the petalinux produces?

Candice-Yin, 2021-09-23:
> Thanks. I've solved this problem in Xilinx SDK. Now I boot my ZC706 board based on the petalinux(http://www.wiki.xilinx.com/Zynq2017.4+Release). I put the image of petalinux in SD card and debug the [...]

hi, would you mind sending me a project of SDK? I'm running this project in SDK using ZCU102 and have some trouble. My mail is 314306956@qq.com. Thank you very much!