dgschwend/zynqnet #44: Can we train your model for zynqnet on any supercomputer cluster with Digits installed on it? What do you suggest?

sara1271, 2018-04-15:
I don't have the required computing resources so I am thinking of getting access to some supercomputer to train the model. I was wondering if your model with datsets can be trained on it if it has DIGITS NVIDIA module available on it?

dgschwend, 2018-04-16:
Hi sara1271,
Zynqnet uses only standard layers, so any (super)computer with Caffe installed should be usable for training. DIGITS itself is not absolutely needed, but very helpful for managing and evaluating the training. Good luck!

sara1271, 2018-04-16:
Thank you for replying back. I am trying to shrink your model as I'm planning on using the PYNQ_Z1 board with Zynqnet which has 630 Block RAM. Any tips or suggestions would be much appreciated.