dgschwend/zynqnet #48: configure the clock rate

denghp5, 2018-05-29:
Hi David, I have read [#2comment] reference as ug585, and configure the clock rate  to 200MHZ,but it didn't work.So I try another way by using the /sys/class/xdevcfg driver to configure the clock rate in zc706,which used your files (setup_fpga.txt). and I transfer setup_fpga.txt to setup_fpga.sh. when I configure the zc706, the UART receive the error:
tee: /sys/device/soc0/amba/f8007000.devcfg/fclk_export: I/O erro.
Can you please explain to me how to that? Thanks a lot.

#!/bin/sh

set -e

write_to () {
    echo $2 | tee $1 > /dev/null
}

set_rate () {
    c=fclk$1
    rate=$2

    d=$(readlink -f "/sys/class/xdevcfg/xdevcfg/device")

    [ -d $d/fclk/$c ] || write_to "$d/fclk_export" $c

    write_to $d/fclk/$c/enable 1
    write_to $d/fclk/$c/set_rate $rate

    echo "Set clock $c to " $(cat $d/fclk/$c/set_rate)
}

# set clock FCLK0 to 125MHz
set_rate 0 200000000

# load Bitstream onto FPGA
loadbit zynq_top.bit

denghp5, 2018-05-30:
Hi, I have tried to run the command "echo fclk0 | xdevcfg/fclk_export" and the terminal responded with"-sh: xdevcfg/fclk_export: Permission denied". Do you have any idea what I can do with that? Thanks.

dgschwend, 2018-05-30:
I don‘t know how to do it for your ZC706 and your Linux/BSP version. You could ask in the Xilinx forums.