dgschwend/zynqnet #54: Error in  C Simulation & Synthesis on vivado_hls 2018.2

ysdagar, 2018-08-19:
Hi David!
Thank you for sharing the great piece of work! 
1. I trying to simulate the code on vivado_hls 2018.2.  & getting the segmentation fault error, right after the computation of first layer is offloaded to FPGA. I have tried increasing SHARED_DRAM to double, as discussed in other issues, however it didn't work out. Please suggest a workaround.

Here is console output  for your reference.

CPU: Network Setup:
=====================

c1    : 256x256 x   3 > 64 , CONV (3x3)/2p + ReLU, IN @mem(       0-  786432B), OUT @mem(  786432B), WEIGHTS @mem(       0-    7168B)
f2/s3 : 128x128 x  64 > 16 , CONV (3x3)/2p + ReLU, IN @mem(  786432- 4980736B), OUT @mem( 4980736B), WEIGHTS @mem(    7168-   44096B)
f2/e1 :  64x64  x  16 > 64 , CONV (1x1)/1  + ReLU, IN @mem( 4980736- 5242880B), OUT @mem( 5242880B), WEIGHTS @mem(   44096-   48448B) (split1)
f2/e3 :  64x64  x  16 > 64 , CONV (3x3)/1p + ReLU, IN @mem( 4980736- 5242880B), OUT @mem( 5243136B), WEIGHTS @mem(   48448-   85568B) (split2)
f3/s1 :  64x64  x 128 > 16 , CONV (1x1)/1  + ReLU, IN @mem( 5242880- 7340032B), OUT @mem( 7340032B), WEIGHTS @mem(   85568-   93824B)
f3/e1 :  64x64  x  16 > 64 , CONV (1x1)/1  + ReLU, IN @mem( 7340032- 7602176B), OUT @mem( 7602176B), WEIGHTS @mem(   93824-   98176B) (split1)
f3/e3 :  64x64  x  16 > 64 , CONV (3x3)/1p + ReLU, IN @mem( 7340032- 7602176B), OUT @mem( 7602432B), WEIGHTS @mem(   98176-  135296B) (split2)
f4/s3 :  64x64  x 128 > 32 , CONV (3x3)/2p + ReLU, IN @mem( 7602176- 9699328B), OUT @mem( 9699328B), WEIGHTS @mem(  135296-  282880B)
f4/e1 :  32x32  x  32 > 128, CONV (1x1)/1  + ReLU, IN @mem( 9699328- 9830400B), OUT @mem( 9830400B), WEIGHTS @mem(  282880-  299776B) (split1)
f4/e3 :  32x32  x  32 > 128, CONV (3x3)/1p + ReLU, IN @mem( 9699328- 9830400B), OUT @mem( 9830912B), WEIGHTS @mem(  299776-  447744B) (split2)
f5/s1 :  32x32  x 256 > 32 , CONV (1x1)/1  + ReLU, IN @mem( 9830400-10878976B), OUT @mem(10878976B), WEIGHTS @mem(  447744-  480640B)
f5/e1 :  32x32  x  32 > 128, CONV (1x1)/1  + ReLU, IN @mem(10878976-11010048B), OUT @mem(11010048B), WEIGHTS @mem(  480640-  497536B) (split1)
f5/e3 :  32x32  x  32 > 128, CONV (3x3)/1p + ReLU, IN @mem(10878976-11010048B), OUT @mem(11010560B), WEIGHTS @mem(  497536-  645504B) (split2)
f6/s3 :  32x32  x 256 > 64 , CONV (3x3)/2p + ReLU, IN @mem(11010048-12058624B), OUT @mem(12058624B), WEIGHTS @mem(  645504- 1235584B)
f6/e1 :  16x16  x  64 > 256, CONV (1x1)/1  + ReLU, IN @mem(12058624-12124160B), OUT @mem(12124160B), WEIGHTS @mem( 1235584- 1302144B) (split1)
f6/e3 :  16x16  x  64 > 256, CONV (3x3)/1p + ReLU, IN @mem(12058624-12124160B), OUT @mem(12125184B), WEIGHTS @mem( 1302144- 1892992B) (split2)
f7/s1 :  16x16  x 512 > 64 , CONV (1x1)/1  + ReLU, IN @mem(12124160-12648448B), OUT @mem(12648448B), WEIGHTS @mem( 1892992- 2024320B)
f7/e1 :  16x16  x  64 > 192, CONV (1x1)/1  + ReLU, IN @mem(12648448-12713984B), OUT @mem(12713984B), WEIGHTS @mem( 2024320- 2074240B) (split1)
f7/e3 :  16x16  x  64 > 192, CONV (3x3)/1p + ReLU, IN @mem(12648448-12713984B), OUT @mem(12714752B), WEIGHTS @mem( 2074240- 2517376B) (split2)
f8/s3 :  16x16  x 384 > 112, CONV (3x3)/2p + ReLU, IN @mem(12713984-13107200B), OUT @mem(13107200B), WEIGHTS @mem( 2517376- 4066112B)
f8/e1 :   8x8   x 112 > 256, CONV (1x1)/1  + ReLU, IN @mem(13107200-13135872B), OUT @mem(13135872B), WEIGHTS @mem( 4066112- 4181824B) (split1)
f8/e3 :   8x8   x 112 > 256, CONV (3x3)/1p + ReLU, IN @mem(13107200-13135872B), OUT @mem(13136896B), WEIGHTS @mem( 4181824- 5215040B) (split2)
f9/s1 :   8x8   x 512 > 112, CONV (1x1)/1  + ReLU, IN @mem(13135872-13266944B), OUT @mem(13266944B), WEIGHTS @mem( 5215040- 5444864B)
f9/e1 :   8x8   x 112 > 368, CONV (1x1)/1  + ReLU, IN @mem(13266944-13295616B), OUT @mem(13295616B), WEIGHTS @mem( 5444864- 5611200B) (split1)
f9/e3 :   8x8   x 112 > 368, CONV (3x3)/1p + ReLU, IN @mem(13266944-13295616B), OUT @mem(13297088B), WEIGHTS @mem( 5611200- 7096448B) (split2)
c10/p1:   8x8   x 736 > 512, CONV (1x1)/1        , IN @mem(13295616-13484032B), OUT @mem(13484032B), WEIGHTS @mem( 7096448- 8605824B) (split1) GLOBAL POOL
c10/p2:   8x8   x 736 > 512, CONV (1x1)/1        , IN @mem(13295616-13484032B), OUT @mem(13486080B), WEIGHTS @mem( 8605824-10115200B) (split2) GLOBAL POOL

CPU: FPGA DRAM Memory Allocation:
     Bytes allocated: 0B (config) + 9878KB (weights) + 13296KB (data)
     region: 47418474455056 – 47418498185360
CPU: Copy Config + Weights to FPGA DRAM:
     0B (config) + 9878KB (weights)
CPU: Loading Input from File ./indata.bin, 768 kBytes.
CPU: Copy Input Data: 768KB (input image)
CPU: Offload CONV Layer c1    : 256x256 x   3 > 64 , CONV (3x3)/2p + ReLU, IN @mem(       0-  786432B), OUT @mem(  786432B), WEIGHTS @mem(       0-    7168B)
FPGA: Computing ................................................................................................................................................................................................................................................................@E Simulation failed: SIGSEGV.
ERROR: [SIM 211-100] CSim failed with errors.

2. I tried to synthesis the same and compilation aborted with a few warnings,

Starting C synthesis ...
/proj/xbuilds/2018.2_daily_latest/installs/lin64/Vivado/2018.2/bin/vivado_hls /wrk/paeg/users/yash/zynqnet/zynqnet_aug18/solution1/csynth.tcl
INFO: [HLS 200-10] Running '/proj/xbuilds/2018.2_daily_latest/installs/lin64/Vivado/2018.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'yashwant' on host 'xhdl4479' (Linux_x86_64 version 2.6.32-642.el6.x86_64) on Sun Aug 19 13:20:00 IST 2018
INFO: [HLS 200-10] On os "Red Hat Enterprise Linux Workstation release 6.8 (Santiago)"
INFO: [HLS 200-10] In directory '/wrk/paeg/users/yash/zynqnet'
INFO: [HLS 200-10] Opening project '/wrk/paeg/users/yash/zynqnet/zynqnet_aug18'.
INFO: [HLS 200-10] Adding design file '_HLS_CODE/fpga_top.cpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/fpga_top.hpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/gpool_cache.cpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/gpool_cache.hpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/image_cache.cpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/image_cache.hpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/memory_controller.cpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/memory_controller.hpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/output_cache.cpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/output_cache.hpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/processing_element.cpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/processing_element.hpp' to the project
INFO: [HLS 200-10] Adding design file '_HLS_CODE/weights_cache.cpp' to the project
[... 136 lines cut ...]
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.3'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.13'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.12'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.11'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.2'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.15'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.14'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.10'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.1'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.0'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.8'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.7'.
WARNING: [ANALYSIS 214-52] Found false inter dependency for variable 'GBRAM.6'.
INFO: [XFORM 203-811] Inferring multiple bus burst write of variable length on port 'memorybus' (_HLS_CODE/memory_controller.cpp:191:14). These data requests might be further partitioned to multiple requests during RTL generation, based on max_read_burst_length or max_write_burst_length settings.
WARNING: [XFORM 203-713] Function 'processInputChannel.' failed dataflow checking:  A dataflow region cannot be instantiated from within a pipelined loop  (_HLS_CODE/processing_element.cpp:47:1), pipe: (_HLS_CODE/image_cache.cpp:70:1). Igno [... cut ...]
  %memorybus_addr_3_wr_s = call i1 @_ssdm_op_WriteReq.m_axi.floatP(float* %memorybus_addr_3, i32 %tmp_46), !dbg !26096
Broken module found, compilation aborted!
Stack dump:
0.	Running pass 'Function Pass Manager' on module '/wrk/paeg/users/yash/zynqnet/zynqnet_aug18/solution1/.autopilot/db/a.o.2.bc'.
1.	Running pass 'Module Verifier' on function '@fpga_top'
/proj/xbuilds/2018.2_daily_latest/installs/lin64/Vivado/2018.2/bin/loader: line 194: 93733 Aborted                 (core dumped) "$RDI_PROG" "$@"
Finished C synthesis.

vishnuchebrolu, 2020-01-31:
Hi @ysdagar ,

WARNING: [HLS 214-114] Only function calls and local variable declarations are allowed in a dataflow region: _HLS_CODE/processing_element.cpp:54:2

Is the above issue fixed?

shawn-wwx, 2020-07-31:
Hi @ysdagar 
I met the same problem as yours. has it been solved yet? would you like to share with me how to solve this?
Thanks a lot!

Taichi-Pink, 2022-02-12:
I solved this problem by commenting out the "pragma hls inline" in "mempry control.cpp" in line 107. If this doesn't work, you can try to comment out other pragmas until the error is gone.