dgschwend/zynqnet #62: Segment fault error == SIGSIVG error

dldldlfma, 2020-04-07:
Hello! I have a question!

When i start the C simulation, error is occur at line number 94  in fpga_top.cpp 

Error message is below.

=============================================================
CPU: Offload CONV Layer c10/p1:   8x8   x 736 > 512, CONV (1x1)/1        , IN @mem(13295616-13484032B), OUT @mem(13484032B), WEIGHTS @mem( 7096448- 8605824B) (split1) GLOBAL POOL
FPGA: Computing .*stopped,reason="end-stepping-range",frame={addr="0x0000000000406d90",func="fpga_top",args=[{name="layer",value="..."},{name="SHARED_DRAM",value="0x38af040"},{name="weights_offset",value="0"},{name="num_weights",value="..."},{name="input_offset",value="2528800"}],file="../../../../fpga_top.cpp",fullname="C:\Users\ksg\Documents\GitHub\zynqnet\_HLS_CODE\fpga_top.cpp",line="101"},thread-id="1",stopped-threads="all"
=============================================================

How can i solve it? 
I don't know how can i go to next step.

when i run the csim.exe in my project path which is below.
C:\Users\ksg\Documents\GitHub\zynqnet\_HLS_CODE\zynqnet_ksg\solution1\csim\build\csim.exe

the error is occur again like below

CPU: FPGA DRAM Memory Allocation:
     Bytes allocated: 0B (config) + 9878KB (weights) + 13296KB (data)
CPU: Copy Config + Weights to FPGA DRAM:
     0B (config) + 9878KB (weights)
CPU: Loading Input from File ./indata.bin, 768 kBytes.
CPU: Copy Input Data: 768KB (input image)
CPU: Offload CONV Layer c1    : 256x256 x   3 > 64 , CONV (3x3)/2p + ReLU, IN @mem(       0-  786432B), OUT @mem(  786432B), WEIGHTS @mem(       0-    7168B)
FPGA: Computing ................................................................................................................................................................................................................................................................ done.
CPU: Offload CONV Layer f2/s3 : 128x128 x  64 > 16 , CONV (3x3)/2p + ReLU, IN @mem(  786432- 4980736B), OUT @mem( 4980736B), WEIGHTS @mem(    7168-   44096B)
FPGA: Computing ................................................................................................................................ done.
CPU: Offload CONV Layer f2/e1 :  64x64  x  16 > 64 , CONV (1x1)/1  + ReLU, IN @mem( 4980736- 5242880B), OUT @mem( 5242880B), WEIGHTS @mem(   44096-   48448B) (split1)
FPGA: Computing ................................................................ done.
CPU: Offload CONV Layer f2/e3 :  64x64  x  16 > 64 , CONV (3x3)/1p + ReLU, IN @mem( 4980736- 5242880B), OUT @mem( 5243136B), WEIGHTS @mem(   48448-   85568B) (split2)
FPGA: Computing ................................................................ done.
CPU: Offload CONV Layer f3/s1 :  64x64  x 128 > 16 , CONV (1x1)/1  + ReLU, IN @mem( 5242880- 7340032B), OUT @mem( 7340032B), WEIGHTS @mem(   85568-   93824B)
FPGA: Computing ................................................................ done.
CPU: Offload CONV Layer f3/e1 :  64x64  x  16 > 64 , CONV (1x1)/1  + ReLU, IN @mem( 7340032- 7602176B), OUT @mem( 7602176B), WEIGHTS @mem(   93824-   98176B) (split1)
FPGA: Computing ................................................................ done.
CPU: Offload CONV Layer f3/e3 :  64x64  x  16 > 64 , CONV (3x3)/1p + ReLU, IN @mem( 7340032- 7602176B), OUT @mem( 7602432B), WEIGHTS @mem(   98176-  135296B) (split2)
FPGA: Computing ................................................................ done.
CPU: Offload CONV Layer f4/s3 :  64x64  x 128 > 32 , CONV (3x3)/2p + ReLU, IN @mem( 7602176- 9699328B), OUT @mem( 9699328B), WEIGHTS @mem(  135296-  282880B)
FPGA: Computing ................................................................ done.
CPU: Offload CONV Layer f4/e1 :  32x32  x  32 > 128, CONV (1x1)/1  + ReLU, IN @mem( 9699328- 9830400B), OUT @mem( 9830400B), WEIGHTS @mem(  282880-  299776B) (split1)
FPGA: Computing ................................ done.
CPU: Offload CONV Layer f4/e3 :  32x32  x  32 > 128, CONV (3x3)/1p + ReLU, IN @mem( 9699328- 9830400B), OUT @mem( 9830912B), WEIGHTS @mem(  299776-  447744B) (split2)
FPGA: Computing ................................ done.
CPU: Offload CONV Layer f5/s1 :  32x32  x 256 > 32 , CONV (1x1)/1  + ReLU, IN @mem( 9830400-10878976B), OUT @mem(10878976B), WEIGHTS @mem(  447744-  480640B)
FPGA: Computing ................................ done.
CPU: Offload CONV Layer f5/e1 :  32x32  x  32 > 128, CONV (1x1)/1  + ReLU, IN @mem(10878976-11010048B), OUT @mem(11010048B), WEIGHTS @mem(  480640-  497536B) (split1)
FPGA: Computing ................................ done.
CPU: Offload CONV Layer f5/e3 :  32x32  x  32 > 128, CONV (3x3)/1p + ReLU, IN @mem(10878976-11010048B), OUT @mem(11010560B), WEIGHTS @mem(  497536-  645504B) (split2)
FPGA: Computing ................................ done.
CPU: Offload CONV Layer f6/s3 :  32x32  x 256 > 64 , CONV (3x3)/2p + ReLU, IN @mem(11010048-12058624B), OUT @mem(12058624B), WEIGHTS @mem(  645504- 1235584B)
FPGA: Computing ................................ done.
CPU: Offload CONV Layer f6/e1 :  16x16  x  64 > 256, CONV (1x1)/1  + ReLU, IN @mem(12058624-12124160B), OUT @mem(12124160B), WEIGHTS @mem( 1235584- 1302144B) (split1)
FPGA: Computing ................ done.
CPU: Offload CONV Layer f6/e3 :  16x16  x  64 > 256, CONV (3x3)/1p + ReLU, IN @mem(12058624-12124160B), OUT @mem(12125184B), WEIGHTS @mem( 1302144- 1892992B) (split2)
FPGA: Computing ................ done.
CPU: Offload CONV Layer f7/s1 :  16x16  x 512 > 64 , CONV (1x1)/1  + ReLU, IN @mem(12124160-12648448B), OUT @mem(12648448B), WEIGHTS @mem( 1892992- 2024320B)
FPGA: Computing ................ done.
CPU: Offload CONV Layer f7/e1 :  16x16  x  64 > 192, CONV (1x1)/1  + ReLU, IN @mem(12648448-12713984B), OUT @mem(12713984B), WEIGHTS @mem( 2024320- 2074240B) (split1)
FPGA: Computing ................ done.
CPU: Offload CONV Layer f7/e3 :  16x16  x  64 > 192, CONV (3x3)/1p + ReLU, IN @mem(12648448-12713984B), OUT @mem(12714752B), WEIGHTS @mem( 2074240- 2517376B) (split2)
FPGA: Computing ................ done.
CPU: Offload CONV Layer f8/s3 :  16x16  x 384 > 112, CONV (3x3)/2p + ReLU, IN @mem(12713984-13107200B), OUT @mem(13107200B), WEIGHTS @mem( 2517376- 4066112B)
FPGA: Computing ................ done.
CPU: Offload CONV Layer f8/e1 :   8x8   x 112 > 256, CONV (1x1)/1  + ReLU, IN @mem(13107200-13135872B), OUT @mem(13135872B), WEIGHTS @mem( 4066112- 4181824B) (split1)
FPGA: Computing ........ done.
CPU: Offload CONV Layer f8/e3 :   8x8   x 112 > 256, CONV (3x3)/1p + ReLU, IN @mem(13107200-13135872B), OUT @mem(13136896B), WEIGHTS @mem( 4181824- 5215040B) (split2)
FPGA: Computing ........ done.
CPU: Offload CONV Layer f9/s1 :   8x8   x 512 > 112, CONV (1x1)/1  + ReLU, IN @mem(13135872-13266944B), OUT @mem(13266944B), WEIGHTS @mem( 5215040- 5444864B)
FPGA: Computing ........ done.
CPU: Offload CONV Layer f9/e1 :   8x8   x 112 > 368, CONV (1x1)/1  + ReLU, IN @mem(13266944-13295616B), OUT @mem(13295616B), WEIGHTS @mem( 5444864- 5611200B) (split1)
FPGA: Computing ........ done.
CPU: Offload CONV Layer f9/e3 :   8x8   x 112 > 368, CONV (3x3)/1p + ReLU, IN @mem(13266944-13295616B), OUT @mem(13297088B), WEIGHTS @mem( 5611200- 7096448B) (split2)
FPGA: Computing ........ done.
**CPU: Offload CONV Layer c10/p1:   8x8   x 736 > 512, CONV (1x1)/1        , IN @mem(13295616-13484032B), OUT @mem(13484032B), WEIGHTS @mem( 7096448- 8605824B) (split1) GLOBAL POOL
FPGA: Computing .....Segmentation fault**

How can i solve it? 

I look forward to your comment.

Thank you!

dgschwend, 2020-04-07:
Have you tried increasing the SHARED_DRAM size as suggested in https://github.com/dgschwend/zynqnet/issues/52?

dldldlfma, 2020-04-07:
Thank you for your response. 

i fixed SHARED_DRAM size in **cpu_top.cpp file at line number 207.**

SHARED_DRAM = (char *)malloc(total_size); 
=>
SHARED_DRAM = (char *)malloc(total_size ** *2 ** );

and it's working!

Thank you so much!

shawn-wwx, 2020-07-31:
hello, guys!
I met the same problem and I tried to change SHARED_DRAM to twice as it was, but it doesn't work as I expected and there is still sigsegv error. 
is there any other possible reason or solution?
Thanks a lot!

ihaterecursion, 2021-01-07:
> hello, guys!
> I met the same problem and I tried to change SHARED_DRAM to twice as it was, but it doesn't work as I expected and there is still sigsegv error.
> is there any other possible reason or solution?
> Thanks a lot!

you can try to delete “blobs/” in fpga_top.cpp