dgschwend/zynqnet #63: How to run the zynqnet on FPGA?

KrisYoung12, 2020-04-08:
Hello, I have successfully generated the cnn ip in HLs and imitated your design to complete the block design. But how do I run the zynqnet on xilinx sdk for windows?

KrisYoung12, 2020-04-08:
@dgschwend I copy all files to SDK, but some files which has this "#include <sys/mman.h>" will report error ( the SDK project is in Windows). How can I resolve it?

ihaterecursion, 2021-01-07:
> Hello, I have successfully generated the cnn ip in HLs and imitated your design to complete the block design. But how do I run the zynqnet on xilinx sdk for windows?
i have some problem in generating the cnn ip in hls，could you please give me some advice？ This is my email：21960204@zju.edu.cn